linuxhw/hw-probe #4: ERROR: failed to upload probe

aadityabagga, 2016-10-24:
Hi,

If I only want to upload hardware related info using the following command, it fails:
~~~~
# hw-probe -probe -upload
Probe for hardware ... Ok

ERROR: failed to upload probe
~~~~

However if I choose to upload all, it gets uploaded:
~~~~
# hw-probe -all -upload
Probe for hardware ... Ok
Reading logs ... Ok
Uploaded to DB, Thank you!

Probe URL:  https://linux-hardware.org/index.php?probe=xxxxxxxxxx
~~~~

Would it possible to only upload the hardware related probe info (in case uploading of logs may contain sensitive information)?

lvc, 2016-10-24:
Hello,

Thank you for reporting this.

Is it OK to upload reduced set of logs including only the following (by -probe option)?

dmidecode
edid
ethtool
hdparm
hwinfo
ifconfig
lsb_release
lsmod
lspci
lspci_all
lsusb
os-release
smartctl
upower
usb-devices
dmesg
xorg.log

All of them except the last two are used to identify the computer and create complete hardware list of devices on board in order to allow people to discover computer's hardware. The last two are used to verify that Linux works on this computer (devices and X11 server are loaded).

Please let me know which of the logs are undesirable for you. Or you just want to upload list of devices and model name of the computer and nothing more?

Thank you.

aadityabagga, 2016-10-25:
Hi Ivc,

I think I would be comfortable in uploading information that does not personally identify my system, for example in the above list ```ifconfig``` could be used to get hardware MAC addresses, and ```xorg.log``` and ```dmesg``` maycontain sensitive information like UUIDs and specific hardware details.

Thanks,
Aaditya

linuxhw, 2018-08-17:
> I think I would be comfortable in uploading information that does not personally identify my system

Since release 1.4 the tool doesn't upload any personal information. Any info uploaded by previous versions has been removed.

See https://github.com/linuxhw/hw-probe#privacy.

linuxhw, 2018-08-17:
Fixed in master.

closed 2018-08-17 by linuxhw