linuxhw/hw-probe #16: Error: ERROR: failed to detect hwaddr

meddario, 2018-02-15:
Hi, on Ubuntu 16.04 (it_IT default locale), i got the error `ERROR: failed to detect hwaddr`, due to the localized output of `ifconfig`. 

To solve this, i had to change the line:

`my $LOCALE = "C.UTF-8";`

to:

`my $LOCALE = "C";`

then the tool worked correctly.

Output of `locale -a` on my system:

```
C
C.UTF-8
en_AG
en_AG.utf8
en_AU.utf8
en_BW.utf8
en_CA.utf8
en_DK.utf8
en_GB.utf8
en_HK.utf8
en_IE.utf8
en_IN
en_IN.utf8
en_NG
en_NG.utf8
en_NZ.utf8
en_PH.utf8
en_SG.utf8
en_US.utf8
en_ZA.utf8
en_ZM
en_ZM.utf8
en_ZW.utf8
it_CH.utf8
it_IT.utf8
POSIX
```

lvc, 2018-02-16:
Fixed by 6226b7b19e19616e212cb2ca5f3f19eec8a770e3.

Thanks a lot!

mikhailnov, 2018-02-16:
Updated git code has been pushed to the hw-probe PPA as hw-probe v. 1.3-9
https://launchpad.net/~mikhailnov/+archive/ubuntu/hw-probe/

closed 2018-08-17 by linuxhw