linuxhw/hw-probe #19: Failed to upload large logs

mikhailnov, 2018-06-01:
```
$ sudo hw-probe -all -upload
Probe for hardware ... Ok
Reading logs ... Ok
ERROR: the probe is larger than 400K
ERROR: failed to upload probe
```
logs/dmesg.1 is >7 MB. 
/root/HW_PROBE/LATEST/ is attached.
I think large logs should be just dropped.

[hw.info.tar.gz](https://github.com/linuxhw/hw-probe/files/2063348/hw.info.tar.gz)

linuxhw, 2018-06-01:
Fixed by the commit 6a1eb65d2098588f0526d3dcc992d8450a5440f2. Please check it.

mikhailnov, 2018-06-01:
Не могу теперь воспроизвести. Огромный dmesg.1 исчез при перезагрузке, скопировал старый большой в /var/log/dmesg.1, но проблема не воспроизвелась, ушла проба https://linux-hardware.org/?probe=ee70294d1d
После обновления пробу проолжают работать

closed 2018-08-17 by linuxhw