linuxhw/hw-probe #21: Privacy vs information brought by some logs

ArchangeGabriel, 2018-06-21:
It looks to me that `blkid`/`fdisk`/`lsblk` as well as `ifconfig` do not provide really interesting data, but otoh gives a lot of information on that computer (like whether FDE is used, are other OSes installed, is a VPN or some other kind of additional network layer is in use).

Is there really a point in collecting those?

ArchangeGabriel, 2018-06-21:
(They are other logs having such issues, I was just listing some)

linuxhw, 2018-10-31:
`blkid` is not collected anymore
`fdisk` output is useful for debugging multi-OS issues on the computer
`lsblk` output is useful for debugging partitions issues on the computer
`ifconfig` is used to check operability of network cards and to get salted sha512 of mac in order to distinguish computers

In 1.5 I'll add an option to skip collecting of unwanted logs, so you'll be able to skip collecting `fdisk` or `lsblk`.

Thanks!

Celti, 2018-11-03:
There are similar privacy implications around systemd information:

Take [systemd-analyze](https://linux-hardware.org/index.php?probe=516ddea8c2&log=systemd-analyze) for an example, which exposes partition UUIDs and device-mapper labels. The [systemctl logs](https://linux-hardware.org/index.php?probe=516ddea8c2&log=systemctl) reveal more of the same.

linuxhw, 2018-12-24:
> There are similar privacy implications around systemd information:
>
> Take [systemd-analyze](https://linux-hardware.org/index.php?probe=516ddea8c2&log=systemd-analyze) for an example, which exposes [...]

Fixed on server. Need to do the same on client side.

Thank you!

linuxhw, 2018-12-24:
> There are similar privacy implications around systemd information:
>
> Take [systemd-analyze](https://linux-hardware.org/index.php?probe=516ddea8c2&log=systemd-analyze) for an example, which exposes [...]

UUIDs do not reveal any private info: https://liquidat.wordpress.com/2013/03/13/uuids-and-linux-everything-you-ever-need-to-know/

linuxhw, 2018-12-27:
> There are similar privacy implications around systemd information:
>
> Take [systemd-analyze](https://linux-hardware.org/index.php?probe=516ddea8c2&log=systemd-analyze) for an example, which exposes [...]

Client-side fix: 10b51160da6a5f9b0ad957d6d79a902e19140d9a

closed 2018-12-27 by linuxhw