linuxhw/hw-probe #24: hw probe fails on curent inxi

smxi, 2018-08-16:
Just FYI, inxi has been rewritten and no longer uses the -! options. Since there are many versions of inxi floating around, and most new ones are the 3.0.x series, which uses --no-host for the the -! 31 option, you probably want to version test inxi then construct the command based on if it's 2.9 or newer or older.

The easiest way to test is to simply read the file, and if the first line contains 'bash', it's old inxi, and if the first line contains 'perl' it's new inxi. Some distros get rid of the /usr/bin/env part so you want to just test for the bash/perl.

sample faliure: https://linux-hardware.org/index.php?probe=0d3c2b71cc&log=inxi
```
 if(check_Cmd("inxi"))
        {
            listProbe("logs", "inxi");
            my $Inxi = runCmd("inxi -! 31 -Fzx -c 0 2>&1");
            writeLog($LOG_DIR."/inxi", $Inxi);
}
```

Just as an aiside, you can get a ton more information than you're getting from inxi.

linuxhw, 2018-08-17:
Fixed by the commit 5baea7bab86437b0717967cd6c957373492bace6.

Thanks a lot!

smxi, 2018-08-17:
The -c 0 is no longer required for inxi 3.x, colors are correctly removed for pipes and redirects, but using -c 0 does not hurt anything. That's just FYI. Note that inxi 3.x gives more information than legacy inxi for -Fxz, more is being added all the time.

closed 2018-08-18 by smxi