linuxhw/hw-probe #25: Collect vainfo for multi-GPU system

mikhailnov, 2018-08-31:
On multi-GPU systemd (notebooks with 2 GPUs) `vainfo` reports information about Intel or another first GPU. I suggest to collect information about other devices.

All devices can be found like this:
```
ls /dev/dri/render*
/dev/dri/renderD128 /dev/dri/renderD129
```
Then query:
`vainfo --display drm --device /dev/dri/renderD128`
`vainfo --display drm --device /dev/dri/renderD129`

Example output:
```
nikolay@Dell-5521:~$ vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_1
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.1 (libva 2.1.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Ivybridge Mobile - 2.1.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
nikolay@Dell-5521:~$ vainfo --display drm --device /dev/dri/renderD129
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/r600_drv_video.so
libva info: Found init function __vaDriverInit_1_1
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.1 (libva 2.1.0)
vainfo: Driver version: mesa gallium vaapi
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
nikolay@Dell-5521:~$ vainfo --device /dev/dri/renderD128
vainfo: unrecognized option '--device'
Show information from VA-API driver
Usage: vainfo --help
	--help print this message

Usage: vainfo [options]
Display options:
	--display display | help         Show information for the specified display, or the available display list 
	--device device                  Set device name, only available under drm display
nikolay@Dell-5521:~$ vainfo --device /dev/dri/renderD129
vainfo: unrecognized option '--device'
Show information from VA-API driver
Usage: vainfo --help
	--help print this message

Usage: vainfo [options]
Display options:
	--display display | help         Show information for the specified display, or the available display list 
	--device device                  Set device name, only available under drm display
nikolay@Dell-5521:~$
```

mikhailnov, 2018-08-31:
DRI_PRIME=1 does not help

linuxhw, 2022-01-17:
Fixed by commit aa36ef4602e2d7fe450679334b6a6b425e6bc038

closed 2022-01-17 by linuxhw

Gomesz785, 2022-01-19:
> Fixed by commit [aa36ef4](https://github.com/linuxhw/hw-probe/commit/aa36ef4602e2d7fe450679334b6a6b425e6bc038)

Hello, so you forget this for 2 years?

linuxhw, 2022-01-21:
> > Fixed by commit [aa36ef4](https://github.com/linuxhw/hw-probe/commit/aa36ef4602e2d7fe450679334b6a6b425e6bc038)
> 
> Hello, so you forget this for 2 years?

Yep(

Gomesz785, 2022-01-22:
> > > Fixed by commit [aa36ef4](https://github.com/linuxhw/hw-probe/commit/aa36ef4602e2d7fe450679334b6a6b425e6bc038)
> > 
> > 
> > Hello, so you forget this for 2 years?
> 
> Yep(

LOL. that is crazy, bud.