linuxhw/hw-probe #31: Convertible category missing on the list

luyatshimbalanga, 2018-12-22:
[linux-hardware.org](https://linux-hardware.org/index.php?view=computers) is missing an item on the list: **convertible**.
See [HP Envy x360 15-cp0xxx](https://linux-hardware.org/?probe=95da2f7884)

linuxhw, 2018-12-24:
All convertibles are in the **Notebook** category for now: https://linux-hardware.org/?view=computers&type=notebook

Not sure if we need a separate category for such devices. Currently one can find convertibles by the `convertible` or `tablet` keyword in the search engine: https://linux-hardware.org/?view=search_computer&computer_model=tablet&computer_type=notebook

Thanks.

linuxhw, 2019-06-07:
Done:

https://linux-hardware.org/?view=computers&type=convertible
https://linux-hardware.org/?view=computers&type=tablet

Also added: "All in one" and "System on chip".

Thanks.

closed 2019-06-07 by linuxhw