linuxhw/hw-probe #33: What does "Failed" without any info mean?

Mannshoch, 2019-06-13:
I tried to Determin the graphicard Problem on my laptop
https://linux-hardware.org/?probe=4cfb07305a

and detected the kernel setup parameter "nomodset" as the source for the problem.

after that change:
https://linux-hardware.org/?probe=b3e69d5f2c
now I only have a "failed" without any hint on this lists.

what could a "failed" without any description mean?

lvc, 2019-06-13:
Hi,

In both cases the `FBDEV` driver is used according to the `Xorg.log`. Free `radeon` driver cannot be used in the `nomodeset` mode.

In the second probe it's marked as Failed due to failure in the `Xorg.log`. The message is empty because `radeon` driver is actually loaded (see `lsmod`). May be the collected X server log is not from the current session? It's usually obtained from `/var/log/Xorg.0.log` or `/home/ubuntu/.local/share/xorg/Xorg.0.log`. May be it is stored in the different location on your system?

Thank you.

Mannshoch, 2019-06-15:
Hi @Ivc

The User `Ubuntu` does not exist on my PC. I was able to locate the log in my home on `/home/$USER/.local/share/xorg/Xorg.0.log`

A colleague told me that my `Core i5 CPU M 430 @ 2.27GHz` has a built in Graphic chip. Which seems missing.

lvc, 2019-06-17:
Your CPU has a built-in graphics, but it's not enabled by your notebook model: http://forum.notebookreview.com/threads/probook-4720s-switchable-graphics.557984/

Do you have `nomodeset` in your `Xorg.0.log` currently?

    grep -nR nomodeset /home/$USER/.local/share/xorg/Xorg.0.log

Mannshoch, 2019-06-17:
No I do not have nomodset in Xorg.0.log. 
After my last post I guessed that may the log could be a Problem and removed all Xorg.0.log and restarted the system.
https://linux-hardware.org/?probe=a6ec431460

I was wondering why I do not have any switch in the BIOS Setup to change the behaviour of the Intel Graphic card. It seems it's really not enabled in this laptop. You may add this information somehow in the list. 

Kind of:
> {HARDWARE} normally has {FEATURE} integrated which seems not supported by this Device.

May also the link to the HW spec from Intel could be interesting.