linuxhw/hw-probe #34: Touchscreen is missing

Mannshoch, 2019-06-18:
Since I found this really nice tool, I try to test all kind of devices with it. Last device I tested was this one:
http://linux-hardware.org/?probe=9e530b2e21
Here is a Touchscreen missing. The touchscreen itself is also not working. 

I use http://q4os.org/ with Trinity because it's the only system I could install on a device with less than 1024 MB RAM. Also special on this device, it has a 32bit EFI but a 64bit CPU.

lvc, 2019-06-19:
Hi,

Looks like you need at least kernel 4.12 and custom firmware to get it visible by the system: https://github.com/TobleMiner/wintron7.0

See https://wiki.debian.org/HowToUpgradeKernel.

But from my point of view the touchscreens on Linux are not so usable like on Android or iOS systems.

Thank you.

Mannshoch, 2020-06-13:
I missed I already requested this. 
https://github.com/linuxhw/hw-probe/issues/62

closed 2020-06-13 by Mannshoch