linuxhw/hw-probe #35: Feature Request: create a Windows Version

Mannshoch, 2019-06-24:
To be able to check if a computer is fully compatible. Please create a Windows Version of this app.

lvc, 2019-06-25:
Thanks for the idea!

Currently you may type your computer model in this form to check for Linux-compatibility: https://linux-hardware.org/?view=search_computer

or run the tool from the Linux LiveCD instead.

lvc, 2019-06-25:
Do you know Windows alternatives to hwinfo or lspci/lsusb command line tools to get lists of ids for PCI/USB/PNP devices?

Mannshoch, 2019-06-25:
Seems possible 

With CMD: https://stackoverflow.com/questions/16445522/how-can-i-get-hardware-ids-serial-numbers-through-command-prompt

With Powershell (In German): https://www.winboard.org/threads/informationen-zur-hardware-aus-dem-system-auslesen.156440/

Mannshoch, 2019-06-25:
By the way. I created a package request for the Android App [Termux](https://github.com/termux/termux-packages/issues/3982)

aadrian, 2019-07-24:
@lvc unfortunately a "LiveCD" (USB) is not even booting on many computers where Windows does :( .

A windows version would be of great help, and way many more users would give it a try.

This would allow e.g. to go to a computer store with an USB Stick to try just try  to find one that could work with Linux :).

alexws54tk, 2019-12-10:
@aadrian
>  unfortunately a "LiveCD" (USB) is not even booting on many computers where Windows does :(

> This would allow e.g. to go to a computer store with an USB Stick to try just try to find one that could work with Linux :).

`То есть загрузиться с флешки в Линукс нам в магазине недадут, а запускать непонятный софт в виндовс дадут? (original text)`

I may run unknown software from USB on windows, but I not may boot from USB to Linux in a computer store? (translated)

Sorry for my bad english.

Mannshoch, 2020-06-18:
If I ask in Computer stores I mostly get the permission to boot from USB.The problem is more that I every time have to tethering to my phone and then download the app.
It's a nice way to test if W-Lan is supportet and sometimes I have to use Bluetooth for connection. 
But it would be way more easy If I'm able to somehow have a live distro which I could preset with Bluetooth and W-Lan settings for my phone and have a link on desktop which I could run.

linuxhw, 2020-07-29:
Because all devices are usually supported under Windows, we don't need to make a classic probe of devices. Instead, we just need to gather simple plain list of PCI and USB ids from the Windows machine and post it to the website with the help of a new option or html form.

aadrian, 2020-07-29:
> I may run unknown software from USB on windows, but I not may boot from USB to Linux in a computer store?

@alexws54tk Yes. It's not that I "may" not be allowed to try to boot, but that it **won't** boot with any Linux live CD/USB, i.e. at boot time will have some incompatibilities, so `hw-probe` can't do it's thing and report back the issues.

> I tested it. Currently all Windows PC in Computerstores do not show a connected USB. So that would not work.

@Mannshoch that depends on the country we are talking about, and also who the store owner trusts :) .

I just want to know which of the computers (if I would buy) from the store would run Linux (or would run with some some minimal tweaks).

That is very valuable for the store owners too, since the manufacturers (except Dell and a few) do not provide this important information.

Mannshoch, 2020-07-29:
At moment, I play around with [Cubic](https://launchpad.net/cubic) to get a live system that boot and automatic runs hw-probe and then present the result in Browser. 
But It seems I'm not pro enough to be able to preset my Mobil phone tethering W-Lan and Bluetooth (if WLAN not work) in the life system and also to directly open the web-page with the result.

sskras, 2020-10-13:
@lvc wrote:

> Do you know Windows alternatives to hwinfo or lspci/lsusb command line tools to get lists of ids for PCI/USB/PNP devices?

One of the ideas is to use portable versions of the utils from Cygwin or Msys2.
**lspci** is directly said to support Cygwin:
http://mj.ucw.cz/sw/pciutils/#:~:text=CYGWIN

... but it has no official package here:
https://cygwin.com/cgi-bin2/package-grep.cgi?grep=lspci

But **lsusb** is here already:
https://cygwin.com/cgi-bin2/package-grep.cgi?grep=lsusb

I don't like Cygwin and prefer it to the modularity and setup-portability of Msys2:
https://superuser.com/questions/1321097/how-to-make-msys-msys2-portable

So the idea is to:
* create a 3 additional packages for Msys2 (import lsusb from Cygwin, import vanilla lspci, and port hw-probe here)
* downstream them to its `MINGW-packages` repository (like [libusb-win32](https://github.com/msys2/MINGW-packages/pull/6679) or other [USB-related packages](https://packages.msys2.org/search?t=pkg&q=usb))
* tell user to install and update Msys2
* tell user to run "pacman -S hw-probe" and then "hw-probe" in the terminal.