linuxhw/hw-probe #38: Flatpak and dmidecode

leinardi, 2019-10-05:
Is `dmidecode` suppose to work with the Flatpak installation?
If I open a shell inside the sandbox with the command `flatpak run --command=sh org.linux_hardware.hw-probe` and I try to run `dmidecode` I get only errors as output:
```
[📦 org.linux_hardware.hw-probe sensors.py]$ dmidecode 
# dmidecode 3.2
/sys/firmware/dmi/tables/smbios_entry_point: Permission denied
Scanning /dev/mem for entry point.
/dev/mem: Permission denied
```
Is this expected?

linuxhw, 2019-10-06:
`sudo` is required to run `dmidecode` (and also some other root stuff like biosdecode, smartctl and hdparm):

    sudo flatpak run org.linux_hardware.hw-probe -all -upload

leinardi, 2019-10-06:
Oh I see, I hoped there was a way to grant the needed permissions with flatpak. Thansk for the clarification.

closed 2019-10-06 by leinardi

linuxhw, 2019-10-06:
I've asked recently about this here: https://github.com/flathub/flathub/issues/775#issuecomment-474395275

leinardi, 2019-10-06:
Flatpak is still convenient as a distribution system, even without the sandbox part. Sad that there is no way to access these kind of resources...