linuxhw/hw-probe #39: how do I remove my sshfs fstab entry from hwprobe on website

Mannshoch, 2019-10-27:
on my hw-probe I have some sshfs entrys in this entries I could see the server with Username.
also hw-probe transfered the # Comments from fstab which contain sensitive Information which I need to modify my sshfs.
please remove all comments on fstab entries

linuxhw, 2019-10-31:
Please report your probe id to info@linux-hardware.org

Thanks!

linuxhw, 2019-10-31:
Found and fixed.

closed 2019-10-31 by linuxhw

Mannshoch, 2019-11-01:
Oh cool. You removed all Comments from all fstab in your database. 
Is hw-probe already updated to not collecting comments ?

linuxhw, 2019-11-02:
> Oh cool. You removed all Comments from all fstab in your database.
> Is hw-probe already updated to not collecting comments ?

Yep: https://github.com/linuxhw/hw-probe/commit/be7fd790e03e3c8fe1e978a652859e45fa3aee57

It will be a part of 1.5.

Thanks.

linuxhw, 2019-11-02:
> Oh cool. You removed all Comments from all fstab in your database.
> Is hw-probe already updated to not collecting comments ?

Moreover, fstab is not collected by default in 1.5.