linuxhw/hw-probe #40: Save the output inside current directory, please

GreatBahram, 2019-11-04:
Hello there,

I think, it would be sensational if you just save the information in current directory rather than root home directory.

Thank you

linuxhw, 2019-12-12:
Hello,

Collected data is stored in the folder with root access only. Do you need a non-removable folder in your current directory?

GreatBahram, 2019-12-21:
Why is that?
I thought you only need `sudo` access in order to get device information, right? I am asking this, because when I want to use stored data, not only I have to use `sudo` command but also need to get the data from `root` home directory. Maybe adding something like `--output` would solve the problem?

linuxhw, 2019-12-21:
> Why is that?

Collected logs may contain private info, so storing them implicitly in the non-root directory is dangerous.

> Maybe adding something like --output would solve the problem?

You have two options here:

1. Use `-save ./DIR` option to save collected archive explicitly in the `DIR` directory
2. Use `-import ./DIR` option to import all your previous probes from server to a local directory

GreatBahram, 2019-12-22:
Thank you very much!

closed 2019-12-22 by GreatBahram