linuxhw/hw-probe #42: hw-probe crashes the system on Rock Pi 4 (RK 3399 aarch64)

aadityabagga, 2019-12-31:
Hi,

I am trying to do a probe of [Rock Pi 4](https://wiki.radxa.com/Rockpi4/hardware/rockpi4).

However when I run `hw-probe`, the system crashes and restarts. This is found in the debug TTL log:

~~~~
[  248.071527] SError Interrupt on CPU5, code 0xbf000000 -- SError
[  248.071529] CPU: 5 PID: 2806 Comm: dmidecode Not tainted 5.3.11-rockchip64 #19.11.3
[  248.071530] Hardware name: Radxa ROCK Pi 4 (DT)
[  248.071530] pstate: 20000000 (nzCv daif -PAN -UAO)
[  248.071531] pc : 0000ffffa82693cc
[  248.071532] lr : 0000aaaaccd06a0c
[  248.071532] sp : 0000ffffd05255f0
[  248.071533] x29: 0000ffffd05255f0 x28: 0000000000000000 
[  248.071535] x27: 0000000000010000 x26: 0000ffffa835a000 
[  248.071536] x25: 0000000000000003 x24: 00000000000f0000 
[  248.071538] x23: 0000000000000000 x22: 0000000000010000 
[  248.071540] x21: 0000aaaaccd20000 x20: 0000aaaad2bbc4a0 
[  248.071541] x19: 0000aaaaccd093b8 x18: 000000000000ffff 
[  248.071543] x17: 0000ffffa82692d0 x16: 0000aaaaccd20e50 
[  248.071544] x15: 0000000000000002 x14: 0000000000000000 
[  248.071546] x13: 0000000000000000 x12: 0000000000000000 
[  248.071547] x11: 0000000000000000 x10: 0000000000000000 
[  248.071548] x9 : 0000000000000000 x8 : 00000000000000de 
[  248.071550] x7 : 0000000000000000 x6 : 0000000000000000 
[  248.071552] x5 : 0000aaaad2bcc4a0 x4 : 0000ffffa836a000 
[  248.071553] x3 : 0000aaaad2bbc4a0 x2 : 0000000000010000 
[  248.071555] x1 : 0000ffffa835a000 x0 : 0000aaaad2bbc4a0 
[  248.071557] Kernel panic - not syncing: Asynchronous SError Interrupt
[  248.071557] CPU: 5 PID: 2806 Comm: dmidecode Not tainted 5.3.11-rockchip64 #19.11.3
[  248.071558] Hardware name: Radxa ROCK Pi 4 (DT)
[  248.071559] Call trace:
[  248.071560]  dump_backtrace+0x0/0x140
[  248.071560]  show_stack+0x14/0x20
[  248.071561]  dump_stack+0xa8/0xd4
[  248.071561]  panic+0x158/0x324
[  248.071562]  __stack_chk_fail+0x0/0x18
[  248.071562]  arm64_serror_panic+0x70/0x80
[  248.071563]  do_serror+0x78/0x158
[  248.071564]  el0_error_naked+0x14/0x1c
[  248.071683] SMP: stopping secondary CPUs
[  248.071684] Kernel Offset: disabled
[  248.071684] CPU features: 0x0002,20006008
[  248.071685] Memory Limit: none

~~~~

Not sure if its a Linux kernel issue or with hw-probe. Tested on [Armbian Buster](https://www.armbian.com/rock-pi-4/) image with Linux 5.3.11. Thanks.

linuxhw, 2019-12-31:
Hi,

Thanks a lot for reporting this.

Looks like this is caused by dmidecode call:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=923988
https://github.com/ARM-software/devlib/issues/315

You can uninstall dmidecode temporarily as a workaround to skip probe of it.

aadityabagga, 2019-12-31:
Thanks, worked after uninstalling `dmidecode`.

Feel free to close the issue as appropriate. :slightly_smiling_face:

closed 2019-12-31 by linuxhw