linuxhw/hw-probe #45: Not all serial numbers are removed

ConiKost, 2020-01-10:
Just grepped over collected logs with latest git version. There still seems to be some serial numbers left? Not all seems to be a real serial number, like the output of hwinfo, which looks like more the device id.

```
arcconf_smart:       serialNumber : 9616EC7B11C22A9752A4E9698A40FC42
arcconf_smart:       serialNumber : 9616EC7B11C22A9752A4E9698A40FC42
cpuid:      PSN: processor serial number           = 569C712DD5E591AE39A4658DA0549DFB
cpuid:   processor serial number: 3B13CE971D617E1BACAEA29774A69559
dmidecode:		Serial services are supported (int 14h)
dmidecode:	Serial Number: To Be Filled By O.E.M.
dmidecode:	Serial Number: 02AE196EAC3931EEE8481F6F1DE5CFE3
dmidecode:	Serial Number: To Be Filled By O.E.M.
dmidecode:	Serial Number: To Be Filled By O.E.M.
dmidecode:	Port Type: Serial Port 16550A Compatible
dmidecode:	Serial Number: 2C2E502AA613C74B027B56D8EB164493
dmidecode:	Serial Number: 79BABC0E0DD270028F24EDA7D6E66F02
dmidecode:	Serial Number: DA0D9094E6284B14EB66F3A22CA06D8C
dmidecode:	Serial Number: 8E3767E52D628F77D6738AC629550B99
dmidecode:	Serial Number: DF3CB255E3B160C7434E693225BF96A7
dmidecode:	Serial Number: A09C3AF45D079106200F3E9A6D680C2E
dmesg:[    9.450196] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
dmesg:[   10.843120] AAC0: serial 0D511189ED3
usb-devices:S:  SerialNumber=0000:00:1a.7
usb-devices:S:  SerialNumber=0000:00:1d.7
usb-devices:S:  SerialNumber=0000:00:1a.0
usb-devices:S:  SerialNumber=0000:00:1a.1
usb-devices:S:  SerialNumber=0000:00:1a.2
usb-devices:S:  SerialNumber=0000:00:1d.0
usb-devices:S:  SerialNumber=0000:00:1d.1
usb-devices:S:  SerialNumber=0000:00:1d.2
hwinfo:  Serial ID: "1BA0887B68CCEF572FD869CC2F673909"
hwinfo:  Serial ID: "0000:00:1d.0"
hwinfo:  Serial ID: "0000:00:1a.0"
hwinfo:  Serial ID: "0000:00:1d.1"
hwinfo:  Serial ID: "0000:00:1a.1"
hwinfo:  Serial ID: "0000:00:1d.2"
hwinfo:  Serial ID: "0000:00:1a.7"
hwinfo:  Serial ID: "0000:00:1a.2"
hwinfo:  Serial ID: "0000:00:1d.7"
smartctl:Serial number:        606D47AE
ioports:  03f8-03ff : serial
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
lsusb:  iSerial                 1 ...
dmi_id:board_serial: 02AE196EAC3931EEE8481F6F1DE5CFE3
arcconf:         Serial number                      : AA08A0B40FD4625FA17C3C5005D5CD0E
arcconf:         Serial number                      : C695CFC16765E492F76994D5845BB2C9
arcconf:         Serial number                      : 83959B62227F332D4416CC5C946FDDE2
arcconf:         Serial number                      : 23770D42D5FD069DDAB7E9626AFD8093
```

linuxhw, 2020-01-11:
Fixed by commit 0c526bb4836709c874f0e5131edec345f07cb875.

Thanks.

closed 2020-01-11 by linuxhw

ConiKost, 2020-01-11:
Updated to latest git. There are some serial numbers still left?
I am not sure. Are there real serial numbers or just some uuid stuff?
If in doubt, drop all them for output?

`
arcconf_smart:       serialNumber : 9616EC7B11C22A9752A4E9698A40FC42
arcconf_smart:       serialNumber : 9616EC7B11C22A9752A4E9698A40FC42
cpuid:      PSN: processor serial number           = 569C712DD5E591AE39A4658DA0549DFB
cpuid:   processor serial number: 3B13CE971D617E1BACAEA29774A69559
dmidecode:              Serial services are supported (int 14h)
dmidecode:      Serial Number: To Be Filled By O.E.M.
dmidecode:      Serial Number: 02AE196EAC3931EEE8481F6F1DE5CFE3
dmidecode:      Serial Number: To Be Filled By O.E.M.
dmidecode:      Serial Number: To Be Filled By O.E.M.
dmidecode:      Port Type: Serial Port 16550A Compatible
dmidecode:      Serial Number: 2C2E502AA613C74B027B56D8EB164493
dmidecode:      Serial Number: 79BABC0E0DD270028F24EDA7D6E66F02
dmidecode:      Serial Number: DA0D9094E6284B14EB66F3A22CA06D8C
dmidecode:      Serial Number: 8E3767E52D628F77D6738AC629550B99
dmidecode:      Serial Number: DF3CB255E3B160C7434E693225BF96A7
dmidecode:      Serial Number: A09C3AF45D079106200F3E9A6D680C2E
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
usb-devices:S:  SerialNumber=...
hwinfo:  Serial ID: "1BA0887B68CCEF572FD869CC2F673909"
smartctl:Serial number:        1BA0887B68CCEF572FD869CC2F673909
ioports:  03f8-03ff : serial
dmi_id:board_serial: 02AE196EAC3931EEE8481F6F1DE5CFE3
arcconf:         Serial number                      : AA08A0B40FD4625FA17C3C5005D5CD0E
arcconf:         Serial number                      : C695CFC16765E492F76994D5845BB2C9
arcconf:         Serial number                      : 83959B62227F332D4416CC5C946FDDE2
arcconf:         Serial number                      : 23770D42D5FD069DDAB7E9626AFD8093
`

linuxhw, 2020-01-11:
These are 32-byte prefixes of salted sha512 hashes of real serials. It's safe to upload them.

Thank you.