linuxhw/hw-probe #47: Wrong info for probe

parkerlreed, 2020-01-24:
https://linux-hardware.org/?probe=d5de6d141a

Corrections:

Year: 2016

DE: Plasma/Kwin

Not sure how those are being detected.

parkerlreed, 2020-01-24:
Plasma is consistently mis-detected as Gnome https://linux-hardware.org/?probe=9fb157fd20

linuxhw, 2020-01-25:
Fixed by 42275b9a7cddd230db2dd775350323d6dbff51f3.

Use `sudo -E hw-probe -all -upload` to properly identify DE.

The manufacture year is detected as date of the first BIOS update. I didn't find a better way. 

Thanks a lot!

closed 2020-01-25 by linuxhw

parkerlreed, 2020-01-25:
Thank you! How are multiple uploads handled from the same hardware?

linuxhw, 2020-01-26:
> Thank you! How are multiple uploads handled from the same hardware?

It's OK to upload multiple probes of the same hardware (say, you need to compare two different states or configurations. e.g to monitor state of hard drives from time to time or test recently added hardware parts).

Computers are identified by a prefix of sha512 hash of MAC address, so multiple probes of the same computer are assigned to one entry in the database (HWid).

But, too frequent duplicated probes are marked for removal in future and are not publicly visible.