linuxhw/hw-probe #49: add support for reporting external monitor info using ddcutil

pabs3, 2020-01-28:
ddcutil speaks the DDC/MCCS protocols for speaking to external VGA/DVI/HDMI monitors. Using the ddcutil detect/capabilities/usbenv/probe/interrogate commands will provide detailed information about the connected monitors.

https://en.wikipedia.org/wiki/Display_Data_Channel
https://www.ddcutil.com/

@rockowitz is the author of ddcutil and would probably appreciate having external monitor DDC information in the https://linux-hardware.org/ database.

PS: please note that the ddcutil output currently contains serial numbers from the monitor EDID data, so those will need to be stripped out.

rockowitz, 2020-01-28:
Command "ddcutil probe" is the command you'd want to use to get 
information about a single monitor.  By default it reports the first 
monitor.  Rather than parsing the output of  "ddcutil detect", it would 
probably be simplest to loop over "ddcutil probe --display N" starting 
at N=1 until failure, or (faster) loop over "ddcutil probe --bus N", for 
N=0..31.

linuxhw, 2020-02-10:
Is it possible to quickly get list of busy i2c bus numbers? Without iterating over all 0..31. It takes 3 seconds on my machine and this doubles execution time of the hw-probe.

rockowitz, 2020-02-10:
Can't think of any way other than opening the /dev/i2c bus and 
attempting to perform an operation. Note that an open may succeed but an 
operation fail with errno EBUSY if another driver is accessing the 
device.  ddcutil queries udev to reduce the set of buses it needs to 
inspect, but that is not the question you're asking.  What are you 
trying to do?

linuxhw, 2020-02-11:
@rockowitz

I need to detect i2c bus numbers where physical monitors are connected. And then run `ddcutil probe --bus N` on them w/o the need to loop through 0..31.

Something like:

    i2cdetect -l | grep "NVIDIA\|nvkm\|i915\|Radeon\|AMDGPU"

On my computer N=2. I need to detect this exact number quickly or somehow reduce the range 0..31.

rockowitz, 2020-02-11:
I think the most efficient way to detect which I2C buses have monitors 
is to examine directories /sys/class/drm/cardN/cardN-*  Directory entry 
edid  returns the EDID if it exists (i.e. if there's a monitor) and 
entry i2c-* gives the I2C bus name.

Note that DDC operations are expensive in terms of elapsed time. ddcutil 
typically spends 90% of it's elapsed time sleeping due to waits mandated 
by the DDC protocol.  The "--stats calls" reports the number of Linux IO 
calls, sleep calls, and the elapsed time for each.

- Sanford

linuxhw, 2020-02-11:
Fixed by f96fd2ff3560cbdec5cea3fd01b58af877ece424. But still need to optimize this and test properly. I have no desktop nearby until Monday (notebooks are not supported).

rockowitz, 2020-02-11:
Correction to my prior post.  I just came across some comments 
indicating that the sysfs i2c-* directory entries do not exist for some 
drivers. However, If an I2C bus does not have a monitor, command 
"ddcutil probe --bus N" is pretty cheap.  It entails a single 50 ms DDC 
protocol mandated sleep, so execution time is just north of that (52 ms 
on the system I'm writing from).

-- Sanford

linuxhw, 2020-02-11:
@rockowitz 

Could you please attach sample output of `ddcutil probe --bus N` for a monitor?

Thanks.

rockowitz, 2020-02-11:
Attached is output from command "ddcutil probe --bus 14"

Some comments:

1) probe is a wrapper for part or all of other commands, particularly 
"ddcutil get vcp scan" and "ddcutil capabilities --verbose", and in turn 
is part of "ddcutil interrogate".  These commands have evolved to 
simplify the instructions given to users who report problems.  As such, 
there are no guarantees as to the stability of the output, though the 
feature scan and capabilities reports have not seen changes in some 
time.  The API, on the other hand, is expected to be  stable.

2) There was a bug in the use of --bus with probe (unexpected code 
path).  I caught it this morning when formulating a reply to you. The 
fix is in the current development branch 0.9.9-dev on github.

-- Sanford

EDID version: 1.4
Mfg id: DEL, model: DELL U3011, sn: PH5NY2CIANXL
Product code: 16485, binary serial number 1095653452 (0x414e584c)

Unsupported feature indicator: DREF_DDC_USES_DDC_FLAG_FOR_UNSUPPORTED

Capabilities for display on bus /dev/i2c-14
Unparsed capabilities string: (prot(monitor)type(lcd)model(U3011)cmds(01 02 03 07 0C E3 F3)vcp(02 04 05 06 08 10 12 14(01 05 08 0B 0C) 16 18 1A 52 60(01 03 04 0C 0F 11 12) AC AE B2 B6 C6 C8 C9 D6(01 04 05) DC(00 02 03 04 05) DF FD)mccs_ver(2.1)mswhql(1))
MCCS version: 2.1
Commands:
   Command: 01 (VCP Request)
   Command: 02 (VCP Response)
   Command: 03 (VCP Set)
   Command: 07 (Timing Request)
   Command: 0c (Save Settings)
   Command: e3 (Capabilities Reply)
   Command: f3 (Capabilities Request)
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 06 (Restore factory geometry defaults)
   Feature: 08 (Restore color defaults)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values (unparsed): 01 05 08 0B 0C
      Values (  parsed):
         01: sRGB
         05: 6500 K
         08: 9300 K
         0b: User 1
         0c: User 2
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 52 (Active control)
   Feature: 60 (Input Source)
      Values (unparsed): 01 03 04 0C 0F 11 12
      Values (  parsed):
         01: VGA-1
         03: DVI-1
         04: DVI-2
         0c: Component video (YPrPb/YCrCb) 1
         0f: DisplayPort-1
         11: HDMI-1
         12: HDMI-2
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B2 (Flat panel sub-pixel layout)
   Feature: B6 (Display technology type)
   Feature: C6 (Application enable key)
[... 293 lines cut ...]
   Feature x70 - Video black level: Blue
   Feature x77 - Unknown feature
   Feature x87 - Sharpness
   Feature x8a - Color Saturation
   Feature x90 - Hue
   Feature x9b - 6 axis hue control: Red
   Feature x9c - 6 axis hue control: Yellow
   Feature x9d - 6 axis hue control: Green
   Feature x9e - 6 axis hue control: Cyan
   Feature x9f - 6 axis hue control: Blue
   Feature xa0 - 6 axis hue control: Magenta
   Feature xa8 - Unknown feature
   Feature xb4 - Source Timing Mode
   Feature xc0 - Display usage time
   Feature xca - OSD
   Feature xcb - Unknown feature
   Feature xcc - OSD Language
   Feature xe3 - Manufacturer Specific
   Feature xfa - Manufacturer Specific
   Feature xfb - Manufacturer Specific
   Feature xff - Manufacturer Specific

Color temperature increment (x0b) = 0 degrees Kelvin
Color temperature request   (x0c) = 2
Requested color temperature = (3000 deg Kelvin) + 2 * (0 degrees Kelvin) = 3000 degrees Kelvin

linuxhw, 2020-02-11:
@rockowitz 

I'd like to add `ddcutil` to the hw-probe AppImage. But I don't see commits in 0.9.9-dev from Feb 11. Did you pushed them?

Thanks.

rockowitz, 2020-02-11:
My oops.  Done.

linuxhw, 2020-02-12:
@rockowitz 

BTW

Tried to add `ddcutil` to Alpine Docker and found missed include of `sys/types.h` in `src/util/edid.h` and other Alpine build incompatibilities: https://git.alpinelinux.org/aports/tree/testing/ddcutil/execinfo.patch

Is it possible to include this patch to upstream?

linuxhw, 2020-02-12:
@rockowitz 

I've tried to build latest 0.9.9 in Alpine 3.11, but failed to link final binary (something is bad with `inline value_bytes_zero`). Is `ddcutil-0.9.1` good enough to use (in Alpine edge)?

rockowitz, 2020-02-12:
What declarations are missing when compiling edid.h for Alpine Linux 
that require sys/types.h?

rockowitz, 2020-02-12:
0.9.1 is almost 2 years old.  Numerous enhancements and bug fixes.  I'd 
rather solve the value_bytes_zero() problem.  What's your 
compiler/linker output?

linuxhw, 2020-02-12:
@rockowitz 

> What declarations are missing when compiling edid.h for Alpine Linux that require sys/types.h?

`ushort` type is missed.

linuxhw, 2020-02-12:
> 0.9.1 is almost 2 years old.  Numerous enhancements and bug fixes.  I'd rather solve the value_bytes_zero() problem.  What's your compiler/linker output?

Full list of problems:

1. Missed `ushort` type in `edid.h`
2. Missed `backtrace_symbols(buffer, nptrs)` in `src/util/debug_util.c`
3. Missed `backtrace(buffer, MAX_ADDRS)` in `src/util/debug_util.c`
4. Missed `sbool` in `src/*/*.c`
5. Undefined reference to `value_bytes_zero`

I've applied a temp fix for 1-4 by `sed`:

    sed -i '1s/^/#include <sys/types.h>/' src/util/edid.h \
    sed -i -e 's/backtrace_symbols(buffer, nptrs)/NULL/' src/util/debug_util.c \
    sed -i -e 's/backtrace(buffer, MAX_ADDRS)/0/' src/util/debug_util.c \
    sed -i -e 's/sbool/bool_repr/g' src/*/*.c

But stuck at 5 with no ideas what is going wrong. The error log of the compiler:

```
...
make[3]: Entering directory '/ddcutil-20200211/src'
  CC       app_ddcutil/ddcutil-main.o
  CC       app_ddcutil/ddcutil-app_dynamic_features.o
  CC       app_ddcutil/ddcutil-app_vcp_info.o
  CC       app_ddcutil/ddcutil-app_dumpload.o
  CC       app_ddcutil/ddcutil-app_setvcp.o
  CC       app_ddcutil/ddcutil-app_getvcp.o
  CCLD     libapp.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     libcommon.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     ddcutil
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: ./.libs/libcommon.a(ddc_vcp.o): in function `ddc_get_nontable_vcp_value':
ddc_vcp.c:(.text+0x1014): undefined reference to `value_bytes_zero'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: ddc_vcp.c:(.text+0x1056): undefined reference to `value_bytes_zero'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:885: ddcutil] Error 1
make[3]: Leaving directory '/ddcutil-20200211/src'
make[2]: *** [Makefile:1132: all-recursive] Error 1
make[2]: Leaving directory '/ddcutil-20200211/src'
make[1]: *** [Makefile:577: all-recursive] Error 1
make[1]: Leaving directory '/ddcutil-20200211'
make: *** [Makefile:484: all] Error 2
```

What I'm trying to do is to add `ddcutil` to the Docker image of `hw-probe` in order to collect DDCs of all connected monitors easily like this was done for `edid-decode`: https://github.com/linuxhw/hw-probe/blob/master/Dockerfile

rockowitz, 2020-02-12:
Interesting.  Normally,  sys/types.h is included from stdlib.h (among 
other places.  There's no harm in adding an explicit include for 
sys/types.h.  I will do that.

linuxhw, 2020-02-12:
@rockowitz

    0.9.1 is almost 2 years old. Numerous enhancements and bug fixes. I'd rather solve the value_bytes_zero() problem. What's your compiler/linker output?

Full list of problems:

1. Missed `ushort` type in `edid.h`
2. Missed `backtrace_symbols(buffer, nptrs)` in `src/util/debug_util.c`
3. Missed `backtrace(buffer, MAX_ADDRS)` in `src/util/debug_util.c`
4. Missed `sbool` in `src/*/*.c`
5. Undefined reference to `value_bytes_zero`

I've applied a temp fix for 1-4 by `sed`:

```
sed -i '1s/^/#include <sys/types.h>/' src/util/edid.h \
sed -i -e 's/backtrace_symbols(buffer, nptrs)/NULL/' src/util/debug_util.c \
sed -i -e 's/backtrace(buffer, MAX_ADDRS)/0/' src/util/debug_util.c \
sed -i -e 's/sbool/bool_repr/g' src/*/*.c
```

But stuck at 5 with no ideas what is going wrong. The error log of the compiler:

```
...
make[3]: Entering directory '/ddcutil-20200211/src'
  CC       app_ddcutil/ddcutil-main.o
  CC       app_ddcutil/ddcutil-app_dynamic_features.o
  CC       app_ddcutil/ddcutil-app_vcp_info.o
  CC       app_ddcutil/ddcutil-app_dumpload.o
  CC       app_ddcutil/ddcutil-app_setvcp.o
  CC       app_ddcutil/ddcutil-app_getvcp.o
  CCLD     libapp.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     libcommon.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     ddcutil
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: ./.libs/libcommon.a(ddc_vcp.o): in function `ddc_get_nontable_vcp_value':
ddc_vcp.c:(.text+0x1014): undefined reference to `value_bytes_zero'
/usr/lib/gcc/x86_64-alpine-linux-musl/9.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: ddc_vcp.c:(.text+0x1056): undefined reference to `value_bytes_zero'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:885: ddcutil] Error 1
make[3]: Leaving directory '/ddcutil-20200211/src'
make[2]: *** [Makefile:1132: all-recursive] Error 1
make[2]: Leaving directory '/ddcutil-20200211/src'
make[1]: *** [Makefile:577: all-recursive] Error 1
make[1]: Leaving directory '/ddcutil-20200211'
make: *** [Makefile:484: all] Error 2
```

What I'm trying to do is to add ddcutil to the Docker image of hw-probe in order to collect DDCs of all connected monitors easily like this was done for edid-decode: https://github.com/linuxhw/hw-probe/blob/master/Dockerfile

rockowitz, 2020-02-12:
>
>     0.9.1 is almost 2 years old. Numerous enhancements and bug fixes.
>     I'd rather solve the value_bytes_zero() problem. What's your
>     compiler/linker output?
>
> Full list of problems:
>
>  1. Missed |ushort| type in |edid.h|
|       As per previous message, will add explicit include of sys/types.h. |
>
>  1. Missed |backtrace_symbols(buffer, nptrs)| in |src/util/debug_util.c|
>  2. Missed |backtrace(buffer, MAX_ADDRS)| in |src/util/debug_util.c|
|configure.ac now checks for the existence of header file execinfo.h.  Type
backtrace code is iftested out if the header file is not found.
|
>
>  1. Missed |sbool| in |src/*/*.c|
>  2. Undefined reference to |value_bytes_zero|
>
> I've applied a temp fix for 1-4 by |sed|:
>
> |sed -i '1s/^/#include <sys/types.h>/' src/util/edid.h \ sed -i -e 
> 's/backtrace_symbols(buffer, nptrs)/NULL/' src/util/debug_util.c \ sed 
> -i -e 's/backtrace(buffer, MAX_ADDRS)/0/' src/util/debug_util.c \ sed 
> -i -e 's/sbool/bool_repr/g' src/*/*.c |
>
> But stuck at 5 with no ideas what is going wrong. The error log of the 
> compiler:
Inline function value_bytes_zero() is defined in ddc_packets.h. inline 
function sbool() is defined in core.h, which is explicitly included by 
just about every source other than those in directory util. Any chance 
inlining is disabled by your compiler?

I could make these functions non-inline, but I'd prefer to limit that 
change to Alpine Linux.  Is there a #define that always gets set?

rockowitz, 2020-02-13:
The latest commits to branch 0.9.9-dev should address your compilation 
and linkage issues.

linuxhw, 2020-02-13:
@rockowitz

Last build error:

```
Making all in .
make[3]: Entering directory '/ddcutil/src'
  CC       app_ddcutil/ddcutil-main.o
  CC       app_ddcutil/ddcutil-app_dynamic_features.o
  CC       app_ddcutil/ddcutil-app_vcp_info.o
  CC       app_ddcutil/ddcutil-app_dumpload.o
  CC       app_ddcutil/ddcutil-app_setvcp.o
  CC       app_ddcutil/ddcutil-app_getvcp.o
  CCLD     libapp.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     libcommon.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     ddcutil
  CC       libmain/api_base.lo
In file included from libmain/api_base.c:33:
./libmain/api_base_internal.h:14:10: fatal error: sys/cdefs.h: No such file or directory
   14 | #include <sys/cdefs.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [Makefile:937: libmain/api_base.lo] Error 1
make[3]: Leaving directory '/ddcutil/src'
make[2]: *** [Makefile:1133: all-recursive] Error 1
make[2]: Leaving directory '/ddcutil/src'
make[1]: *** [Makefile:578: all-recursive] Error 1
make[1]: Leaving directory '/ddcutil'
make: *** [Makefile:485: all] Error 2
```

`/usr/include/sys/cdefs.h` is provided only by `bsd-compat-headers` package on Alpine with a deprecation warning for `sys/cdefs.h`.

rockowitz, 2020-02-13:
I've deleted the include of sys/cdefs.h and pushed the change to 
github.  It does not appear to be necessary.  If it had been, how would 
you have proposed dealing with this Alpine Linux deviation.

linuxhw, 2020-02-13:
@rockowitz 

> I've deleted the include of sys/cdefs.h and pushed the change to github.  It does not appear to be necessary.  If it had been, how would you have proposed dealing with this Alpine Linux deviation.

Probably not all apps are compatible with it due to different libc implementation. But it rocks when creating portable Docker apps due to extremely light weight base system.

linuxhw, 2020-02-13:
@rockowitz 

> I've deleted the include of sys/cdefs.h and pushed the change to github.  It does not appear to be necessary.  If it had been, how would you have proposed dealing with this Alpine Linux deviation.

Seems that this include was necessary for `__THROW` in `api_base_internal.h`. I see a bunch of related build errors currently.

linuxhw, 2020-02-13:
@rockowitz 

> I've deleted the include of sys/cdefs.h and pushed the change to github.  It does not appear to be necessary.  If it had been, how would you have proposed dealing with this Alpine Linux deviation.

Could you please revert this change? I'll try to use `bsd-compat-headers` to get `sys/cdefs.h` as a temp solution.

Thanks.

linuxhw, 2020-02-13:
@rockowitz 

Please look: https://wiki.musl-libc.org/faq.html#Q:-When-compiling-something-against-musl,-I-get-error-messages-about-%3Ccode%3Esys/cdefs.h%3C/code%3E

rockowitz, 2020-02-13:
Builds cleanly for me.  Likely because sys/types.h is pulled in 
indirectly.   A quick scan of /usr/include show's it's actually hard to 
avoid.  Alpine Linux must surely already have a way to deal with this.

rockowitz, 2020-02-13:
Ignore the prior message re sys/types.h, not sys/cdef.h.  I need to take 
care of some things before the snowstorm hits.  I'll look at this later.

rockowitz, 2020-02-14:
I have modified files api_base_internal.h and api_base.c to, insofar as 
I can tell, not depend on cdefs.h by replacing the __THROW macro with 
the C code it would expand to in the current context. Let me know if 
this works for you.  If it does, cleanup will follow.

linuxhw, 2020-02-14:
> I have modified files api_base_internal.h and api_base.c to, insofar as I can tell, not depend on cdefs.h by replacing the __THROW macro with the C code it would expand to in the current context. Let me know if this works for you.  If it does, cleanup will follow.

Build succeeded on 9f320. Thanks a lot!

Started build of d378f.

linuxhw, 2020-02-14:
@rockowitz 

> Build succeeded on 9f320. Thanks a lot!
>
> Started build of d378f.

Build succeeded on d378f.

linuxhw, 2020-02-14:
Built `1.5-ddc` tag of https://hub.docker.com/r/linuxhw/hw-probe for testing purposes:

```
xhost +local:
sudo docker run -it -v /dev:/dev:ro -v /lib/modules:/lib/modules:ro -v /etc/os-release:/etc/os-release:ro -v /var/log:/var/log:ro --privileged --net=host --pid=host linuxhw/hw-probe:1.5-ddc -all -upload
```

Also AppImage is ready: https://github.com/linuxhw/hw-probe/releases/download/1.5/hw-probe-1.5-159-x86_64.AppImage

    chmod +x ./hw-probe-1.5-159-x86_64.AppImage
    sudo ./hw-probe-1.5-159-x86_64.AppImage -all -upload

I'll test them when I access my desktop on Monday.

linuxhw, 2020-02-17:
@rockowitz 

Got first probe of Dell P2414H by Docker image: https://linux-hardware.org/?probe=546f54f7e3&log=ddcutil

But it takes about 40 seconds to `ddcutil probe --bus 6` on my desktop. Is it possible to speed up somehow? Where is the bottleneck?

I want to enable this probe by default to create large database of `DDC` probes, but execution time should be limited to 5 sec max per monitor.

Thanks.

linuxhw, 2020-02-17:
@rockowitz 

Tried to run on another desktop, but `ddcutil` can't find a monitor (DELL via HDMI on Fedora 31 X11): https://github.com/rockowitz/ddcutil/issues/106

linuxhw, 2020-02-17:
> @rockowitz
> 
> Tried to run on another desktop, but `ddcutil` can't find a monitor (DELL via HDMI on Fedora 31 X11): [rockowitz/ddcutil#106](https://github.com/rockowitz/ddcutil/issues/106)

Fixed by `xhost +local:`

linuxhw, 2020-02-17:
Second probe (DELL U2417H): https://linux-hardware.org/?probe=9e951ac165&log=ddcutil

rockowitz, 2020-02-17:
Unfortunately the long execution time is a consequence of the DDC 
protocol spec, which mandates that the host wait 50-200 ms after writing 
a command to the display before requesting a reply.  This is compounded 
for displays with marginal I2C implementations that require retries.  
ddcutil spends approximately 90% of its time in sleeps mandated by the 
DDC protocol. Take a look at the output produced by the --stats option, 
which reports both the time spent sleeping and the number of I2C retries 
required.

The sleep time can be adjusted using the -- sleep-multiplier option, 
which takes a floating point number as an argument. Values < 1 reduce 
the sleep times from those in the spec, values > 1 increase it.   For 
example, the Dell U3011 in front of me works reliably with a 
sleep-multiplier of .1!  On the other hand, communication with the Dell 
P2411h in my multi-monitor setup requires a sleep-multiplier in the 2-4 
range for reliable operation.  (I keep it in the testing mix precisely 
because its DDC implementation is so marginal.) You might improve 
performance overall by defaulting to a low sleep-multiplier value, then 
retrying with successively greater values in case of failure.

It is conceivable that the sleep-multiplier could be dynamically 
adjusted for long running commands,  increasing or decreasing depending 
on error rates. But that, as they say, is a project, and I'm not sure 
how much performance would be achieved overall.

--Sanford

rockowitz, 2020-02-18:
I've made some experimental changes to reduce execution time. The 
changes have been pushed to branch 0.9.9-dev.  One that appears to not 
introduce errors, but needs much more testing, eliminates certain types 
of sleep events.  For now, it is enabled using undocumented option 
"--f1".  Give it a try.

linuxhw, 2020-02-18:
> The sleep time can be adjusted using the -- sleep-multiplier option

    sudo ddcutil probe --bus 6
    # Takes 37.4s

    sudo ddcutil probe --bus 6 --sleep-multiplier 0.1 | grep -v "DDC Null Response"
    # Takes 15.0s

Output is not changed.

Is there an option to hide `ddc_write_read() succeeded after N sleep and retry ...` and `Extended delay as recovery ...` messages?

linuxhw, 2020-02-18:
> I've made some experimental changes to reduce execution time. The changes have been pushed to branch 0.9.9-dev.  One that appears to not introduce errors, but needs much more testing, eliminates certain types of sleep events.  For now, it is enabled using undocumented option "--f1".  Give it a try.

    sudo ddcutil-prev probe --bus 6
    # Takes 37.4s

    sudo ddcutil-now probe --bus 6
    # Takes 20.8s

    sudo ddcutil-now probe --bus 6 --sleep-multiplier 0.1
    # Takes 16.8s

Output is not changed. Added `set_error_stats_sleep_multiplier_factor` value to the output.

Thanks!

linuxhw, 2020-02-18:
@rockowitz

Does the best `sleep-multiplier` value depend on cpu or monitor itself?

I got best performance on `sleep-multiplier=0.2` on Dell P2414H:

    sudo ddcutil-now probe --bus 6 --f1 --sleep-multiplier 0.2
    # Takes 8.5s

rockowitz, 2020-02-18:
All normal sleep times, i.e. as specified by the DDC spec, are 
multiplied by the --sleep-multiplier value.  So a normal 50 ms wait 
before querying a response from the monitor becomes 10 ms.  Success 
depends on the monitor only in the sense that some monitors can compose 
responses in the reduced time, some cannot.   --sleep-multiplier is 
useful when targeting a specific monitor whose delay sensitivity is 
known.  Applied to an arbitrary monitor, a given --sleep-multiplier may 
work, or it may not.

rockowitz, 2020-02-18:
> The sleep time can be adjusted using the -- sleep-multiplier option
>
> |sudo ddcutil probe --bus 6 # Takes 37.4s sudo ddcutil probe --bus 6
> --sleep-multiplier 0.1 | grep -v "DDC Null Response" # [...]
This debugging message is presently always on so that use of specialized 
recovery code for handling the DDC Null Message is apparent.  It will be 
disabled before release 0.9.9 is published.

linuxhw, 2020-02-19:
> I've made some experimental changes to reduce execution time. The changes have been pushed to branch 0.9.9-dev.  One that appears to not introduce errors, but needs much more testing, eliminates certain types of sleep events.  For now, it is enabled using undocumented option "--f1".  Give it a try.

Can I use `--f1` in production? Will it be renamed in future?

rockowitz, 2020-02-19:
> I've made some experimental changes to reduce execution time. The
>     changes have been pushed to branch 0.9.9-dev. One that appears to
>     not introduce errors, but needs much more testing, [...]
Options named --f1, --i1 etc. are undocumented and intended testing 
purposes.  Their meaning can change at any time.   If the changes are 
determined not to introduce errors they will simply become part of 
normal execution.

pabs3, 2020-02-22:
I wonder if making hw-probe run multiple processes in the background
and aggregating their output would mitigate this issue by having the
tool runtime be parallel instead of serial.

linuxhw, 2020-02-22:
> I wonder if making hw-probe run multiple processes in the background and aggregating their output would mitigate this issue by having the tool runtime be parallel instead of serial.

@rockowitz 

Can a monitor handle multiple ddc requests in parallel?

rockowitz, 2020-02-23:
>
>     I wonder if making hw-probe run multiple processes in the
>     background and aggregating their output would mitigate this issue
>     by having the tool runtime be parallel instead of serial.
You should be able to run multiple copies of ddcutil in parallel, so 
long as each accesses only a single I2C device, i.e. uses the --bus option.

It is possible to access multiple displays simultaneously using a single 
copy of the shared library, with a separate thread for each display.   
This is what ddcui does.

> @rockowitz <https://github.com/rockowitz>
>
> Can a monitor handle multiple ddc requests in parallel?
No.

pabs3, 2020-02-27:
To be clear, I meant running ddcutil in parallel with other tools for hardware info extraction, but I agree that running ddcutil itself in parallel would be nice.

closed 2020-07-29 by linuxhw