linuxhw/hw-probe #50: Hardware probe says I use Manjaro XFCE, but I'm running KDE

dginovker, 2020-02-02:
I've installed XFCE in the past, but it's been a long time since I've used it. Here's a link to my probe: https://linux-hardware.org/?probe=86410e30f4

My `neofetch`:

```
dr_bracket@dr-pc ~/Downloads $ neofetch 
██████████████████  ████████   dr_bracket@dr-pc 
██████████████████  ████████   ---------------- 
██████████████████  ████████   OS: Manjaro Linux x86_64 
██████████████████  ████████   Kernel: 5.4.14-2-MANJARO 
████████                         ████████   Uptime: 2 hours, 4 mins 
████████  ████████  ████████   Packages: 1786 (pacman), 9 (flatpak) 
████████  ████████  ████████   Shell: bash 5.0.11 
████████  ████████  ████████   Resolution: 1920x1080 
████████  ████████  ████████   DE: Plasma 
████████  ████████  ████████   WM: KWin 
████████  ████████  ████████   Theme: Breath [GTK3] 
████████  ████████  ████████   Icons: maia [GTK3] 
████████  ████████  ████████   Terminal: konsole 
████████  ████████  ████████   Terminal Font: Noto Mono 10 
                               CPU: Intel i5-6400 (4) @ 3.300GHz 
                               GPU: Intel HD Graphics 530 
                               Memory: 2234MiB / 15912MiB 
```

linuxhw, 2020-02-02:
Hello,

This is now fixed. Please use

    sudo -E hw-probe -all -upload

for better identifying of the DE.

Thanks a lot!

closed 2020-02-02 by linuxhw

grahamperrin, 2021-04-02:
Smart! I would never have guessed. 

@linuxhw aside from the DE, is there any other aspect that's truer with sudo as the logged in user, than as root _without_ sudo?

linuxhw, 2021-04-02:
AFAIK, DE is the only attribute that requires `sudo -E`.