linuxhw/hw-probe #52: POP!_OS is detected as Ubuntu.

aadrian, 2020-02-11:
POP!_OS seems to be detected as Ubuntu by HW-Probe.

Even if POP!_OS is based on Ubuntu, it does more and IMHO should be recognized separately.

 On quite a few systems Ubuntu 19.10 won't even boot, but a POP!_OS version (based on Ubuntu 19.10 will have no issues booting and installing.

Thank you.

linuxhw, 2020-02-17:
Agree.

We support Pop!_OS: https://linux-hardware.org/?distro=Pop_OS

Please let us know your probe ID to debug why it's not recognized.

Thanks.

aadrian, 2020-02-17:
@linuxhw e.g. this probe https://linux-hardware.org/index.php?probe=96549aeaf8 also exhibits this issue. 

From the boot log https://linux-hardware.org/index.php?probe=96549aeaf8&log=boot_efi is visible that it's POP!_OS, not Ubuntu)

linuxhw, 2020-02-18:
Fixed by commit 48685f7197f04d3ad9e4f8df337dbeb358674b62. Thank you.

closed 2020-02-18 by linuxhw