linuxhw/hw-probe #53: Can’t run hw-probe

kocoman2, 2020-02-13:
I am in ubuntu 19.10 and get this error

Executing hw-probe -all -upload

ERROR: failed to detect Linux distribution

ERROR: Make sure required Snap interfaces are connected:

    for i in hardware-observe system-observe block-devices log-observe upower-observe physical-memory-observe network-observe raw-usb mount-observe opengl;do sudo snap connect hw-probe:$i :$i; done
root@k:~# 

Thanks

linuxhw, 2020-02-17:
Hi,

What happens after connecting these interfaces?

Please execute the above command in your terminal and re-run the `hw-probe -all -upload`.

Thanks.

linuxhw, 2020-03-30:
Also please try https://packages.ubuntu.com/focal/hw-probe instead of snap.

See https://wiki.ubuntu.com/Hardware_probe

closed 2020-07-29 by linuxhw