linuxhw/hw-probe #55: Failed to detect unknown camera on Toshiba U920T

Brawlence, 2020-02-21:
Ubuntu 18.04.4 LTS
**hw-probe** results: https://linux-hardware.org/index.php?probe=d87dadfcd6
local **lshw** report:

id: | usb:2
-- | --
description: | Video
product: | TOSHIBA Web Camera - HD
vendor: | Chicony Electronics Co.,Ltd.
physical id: | 3
bus info: | usb@1:1.3
version: | 60.37
serial: | 0x0001
capabilities: | usb-2.00
configuration: | driver=uvcvideo maxpower=500mA speed=480Mbit/s | driver | = | uvcvideo | maxpower | = | 500mA | speed | = | 480Mbit/s

id: | usb:3
-- | --
description: | Video
product: | USB 2.0 PC Camera
vendor: | Alcor Micro, Corp.
physical id: | 4
bus info: | usb@1:1.4
version: | 1.00
capabilities: | usb-2.00
configuration: | driver=uvcvideo maxpower=500mA speed=480Mbit/s | driver | = | uvcvideo | maxpower | = | 500mA | speed | = | 480Mbit/s

The USB 2.0 PC Camera (Alcor) is some-why not listed as a device in the report, although it detects the Toshiba one fine

linuxhw, 2020-02-21:
Could you please rerun probe by https://packages.ubuntu.com/focal/hw-probe ?

http://ftp.debian.org/debian/pool/main/h/hw-probe/hw-probe_1.4-3_all.deb

Probably this is a bug in Snap itself.

Thank you for reporting this.

Brawlence, 2020-02-21:
Here's [result for the PPA](https://linux-hardware.org/?probe=2337006eb5):  
```
brawlence@XXXX:~$ sudo apt-get update | grep "hw-probe"
Hit:14 http://ppa.launchpad.net/mikhailnov/hw-probe/ubuntu bionic InRelease
^C
brawlence@XXXX:~$ sudo -E hw-probe
Executing hw-probe -all -upload

Probe for hardware ... Ok
Reading logs ... Ok
WARNING: failed to create EDID
Uploaded to DB, Thank you!

Probe URL: https://linux-hardware.org/?probe=2337006eb5
```

Here's one [for .deb](https://linux-hardware.org/?probe=f2afb7ef64):
```

brawlence@XXXX:~$ sudo apt show hw-probe | grep "Sources"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

APT-Sources: /var/lib/dpkg/status
brawlence@XXXX:~$ sudo -E hw-probe -all -upload -id "this was done by a deb package"
Probe for hardware ... Ok
Reading logs ... Ok
WARNING: failed to create EDID
Uploaded to DB, Thank you!

Probe URL: https://linux-hardware.org/?probe=f2afb7ef64

```

Brawlence, 2020-02-21:
Huh, strange. I don't know what happened (I haven't done anything related), but Ubuntu had seemingly fetched the driver and now both **lshw** and **hw-probe** are reporting the `Alcor Micro camera` as `Toshiba Web Camera - 3M`:

id: | usb:2
-- | --
description: | Video
product: | TOSHIBA Web Camera - HD
vendor: | Chicony Electronics Co.,Ltd.
physical id: | 3
bus info: | usb@1:1.3
version: | 60.37
serial: | 0x0001
capabilities: | usb-2.00
configuration: | driver=uvcvideo maxpower=500mA speed=480Mbit/s | driver | = | uvcvideo | maxpower | = | 500mA | speed | = | 480Mbit/s

id: | usb:3
-- | --
description: | Video
product: | TOSHIBA Web Camera - 3M
vendor: | Alcor Micro, Corp.
physical id: | 4
bus info: | usb@1:1.4
version: | 38.75
capabilities: | usb-2.00
configuration: | driver=uvcvideo maxpower=500mA speed=480Mbit/s | driver | = | uvcvideo | maxpower | = | 500mA | speed | = | 480Mbit/s

The issue seems to be resolved, thank you.

closed 2020-02-21 by Brawlence