linuxhw/hw-probe #59: Package 'dmidecode' and 'lscpu' not available for FreeBDS on arm

tschuxxi, 2020-05-23:
It seems like the package the packages dmidecode and lscpu are not available when running FreeBSD on ARM. This means the install script is not working.

linuxhw, 2020-05-24:
Fixed by commit https://github.com/linuxhw/hw-probe/commit/17eccc711b8336036659fd9c0eb98b7fefe8a450

Thank you!

danfe, 2020-05-29:
Nice.  However, it still creates those files, just zero-sized.  For example, on FreeBSD/powerpc (Mac mini G4), I see three empty files: `dmidecode`, `lspci`, and `lsusb`.  Perhaps the fix can be extended to only keep files with actual contents.

linuxhw, 2020-05-30:
> Nice. However, it still creates those files, just zero-sized. For example, on FreeBSD/powerpc (Mac mini G4), I see three empty files: `dmidecode`, `lspci`, and `lsusb`. Perhaps the fix can be extended to only keep files with actual contents.

This is OK. This is intended to let the user know that such information could potentially be collected. Both `lspci` and `lsusb` can be installed and collected. `dmidecode` can be ported to arm and collected: http://ftp.netbsd.org/pub/pkgsrc/current/pkgsrc/sysutils/dmidecode/README.html

closed 2020-07-29 by linuxhw