linuxhw/hw-probe #60: Probe for hardware stuck

drillsar, 2020-06-01:
I have Gentoo and installed it; I have kernel 5.6.14 if that makes a difference. It just says Probe for hardware ... and does nothing after that.

bsdhw, 2020-06-01:
Hi,

Could you please run it with additional `-list` option to list all executed commands?

This will let us to know which command hanged.

Thanks.

drillsar, 2020-06-03:
` hw-probe --all --upload --list
logs/os-release
logs/system-release
logs/lsb-release
Executing: uname -m
Executing: uname -r
logs/dmi_id
logs/ifconfig
Executing: ifconfig -a 2>&1
Probe for hardware ... 
logs/dev
Executing: find /dev -ls 2>/dev/null
logs/lsmod
Executing: lsmod 2>&1
logs/hwinfo
Executing: hwinfo --monitor --bluetooth --bridge --camera --cdrom --chipcard --cpu --disk --dvb --fingerprint --floppy --gfxcard --hub --ide --isapnp --isdn --joystick --keyboard --mouse --netcard --network --pci --pcmcia --scanner --scsi --smp --sound --tape --tv --usb --usb-ctrl --wlan 2>/dev/null
`

bsdhw, 2020-06-03:
Could you please run this command separately? Is it hangs?

If so, could you please try to identify hanged option by bisecting the list of options?

Probably one of the following options hangs:

    dvb
    floppy
    isapnp
    isdn
    joystick
    netcard
    smp
    tape

Thanks a lot for your help.

drillsar, 2020-06-03:
How do you run separate? Can you please explain?

bsdhw, 2020-06-04:
Just run this command in terminal:

    hwinfo --monitor --bluetooth --bridge --camera --cdrom --chipcard --cpu --disk --dvb --fingerprint --floppy --gfxcard --hub --ide --isapnp --isdn --joystick --keyboard --mouse --netcard --network --pci --pcmcia --scanner --scsi --smp --sound --tape --tv --usb --usb-ctrl --wlan

drillsar, 2020-06-07:
I did that and get the following bios.4.3: ddc info but doesn't do anything after that.

bsdhw, 2020-06-17:
Could you please repeat it with `sudo`?

Thanks.

linuxhw, 2020-07-29:
@drillsar 

Could you please try:

    sudo -E hwinfo --monitor --bluetooth --bridge --camera --cdrom --chipcard --cpu --disk --dvb --fingerprint --floppy --gfxcard --hub --ide --isapnp --isdn --joystick --keyboard --mouse --netcard --network --pci --pcmcia --scanner --scsi --smp --sound --tape --tv --usb --usb-ctrl --wlan

Thanks.

linuxhw, 2020-12-10:
Please provide information.

fieryrisephoenix, 2024-04-05:
Hi, I have the same problem. I'm getting stuck on the --cd rom. If you remove it, then everything is fine

rvdbergh, 2024-10-15:
Same issue here on my 2012 Asus K95VJ laptop that I SSH into running Debian 12, it gets stuck on --usb only reporting usb.2: usb

sebma, 2025-01-13:
On my system, it hangs because of `inxi -F`