linuxhw/hw-probe #61: Please use /var/run/dmesg.boot instead of dmesg(8) output (or just both).

vermaden, 2020-06-09:
Hi,

after 30+ days of uptime on my laptop the only thing you will get from **dmesg(8)** command are mostly this:

```
wlan0: link state changed to DOWN
wlan0: link state changed to UP
wlan0: link state changed to DOWN
wlan0: link state changed to UP
wlan0: link state changed to DOWN
wlan0: link state changed to UP
wlan0: link state changed to DOWN
```

Thus using (or at least adding) the contents of the **/var/run/dmesg.boot** file would be useful.

Regards,
vermaden

pabs3, 2020-07-11:
On systemd based systems the full dmesg log (unless it rotated) for the current boot can be obtained like this:

```
sudo journalctl --dmesg
```

vermaden, 2020-07-11:
> On systemd based systems the full dmesg log (unless it rotated) for the current boot can be obtained like this:
> 
> ```
> sudo journalctl --dmesg
> ```

I was saying about FreeBSD which does not have systemd.

bsdhw, 2020-07-29:
Fixed by commit 3495752099333da2f27abc02f3d5b68151a20574.

Thank you!

closed 2020-07-29 by linuxhw