linuxhw/hw-probe #63: Unity isn't detected

Recydywa, 2020-06-15:
I'm using Unity as my DE but hw-probe reported GNOME (X11) - GDM (with .deb) and no DE (with snap). Also in both cases year is reported as 2019.

I first installed the .deb package on Ubuntu 20.04 LTS. Then I tried snap package from edge channel. When I check version they both report as 1.5

Recydywa, 2020-06-16:
Hello,

I've launched both (.deb and snap) from command line.

`sudo snap run hw-probe -all -upload`

https://linux-hardware.org/index.php?probe=da8da027bf  (snap)

https://linux-hardware.org/index.php?probe=6e7f34f58c  (.deb)

Thank you for quick response.

bsdhw, 2020-06-17:
You need to use `sudo -E` instead of `sudo` to detect DE properly in both cases.

Recydywa, 2020-06-17:
So in case of .deb I did what was posted on Ubuntu wiki.

https://wiki.ubuntu.com/Hardware_probe

`sudo -E hw-probe -all -upload`

I did the same today and it did correctly detect Unity. Not sure why it didn't before.

Also the snap package detected Unity so it seems it was a false alarm.

Thanks for your help.

closed 2020-06-17 by Recydywa