linuxhw/hw-probe #66: Enhance search on Website

Mannshoch, 2020-07-05:
Enhance search on Website with better Filter possibilities.

See Mockup
[image]

bsdhw, 2020-07-06:
All filters are implemented on this page: https://linux-hardware.org/?view=trends

Filter by screen type is not implemented yet because we don't know how to detect it. Do you have an idea?

Thanks.

Mannshoch, 2020-07-06:
I know that filtering is available, but I really could not use it. It seems more like a sort of stats view.
The current View and behavior is more a view for experts and not for normal users like me.

What's with your comment here?
https://github.com/linuxhw/hw-probe/issues/62#issuecomment-645560196
Didn't that work? 

Please see the Mockup more as a kind of Example how you could work with and add new filter. Not what Filter have to be available.

Mannshoch, 2020-09-29:
Searching for compatibility is really annoying. e.g. if you search at first for a HP device but think after that about filtering for the year. the Vendor setting get reset. 
Also difficult. If you pined down some Models that may could be a what you search. There is no possibility to compare them to detect the big differences.
Comparing itself is very difficult. because you may have found a device in your favorite store but you could not find it in the table. what would you do. It would be nice If I could find the most problematic parts among some devices.
I may also currious abot the Hardware evolving over some years betweens some models.  
e.g. "show me all 'HP ENVY x360 Convetible*' and their diferences in compatibility" 
or "show me the CPU performance evolving over the models the last five years"

grahamperrin, 2022-03-14:
From opening post <https://github.com/linuxhw/hw-probe/issues/66#issue-651101353>: 

> … Filter …

Also e.g. #112

Mannshoch, 2023-05-05:
Currently, I'm looking for an OLD AMD Graphic card.
I have requirements:

1. AMD
2. Compatible with Secureboot -> GOP 
3. Two HDMI Ports

It would be nice If I could search for such feature.