linuxhw/hw-probe #68: Appimage run failed

Surendrajat, 2020-08-07:
Logs:
```bash
$ sudo -E ./hw-probe-1.5-149-x86_64.AppImage -all -upload                                      ✔ 

[sudo] password for user:          
QStandardPaths: wrong ownership on runtime directory /run/user/1000, 1000 instead of 0
QStandardPaths: wrong ownership on runtime directory /run/user/1000, 1000 instead of 0

(AppImageLauncher:26512): GLib-GIO-CRITICAL **: 18:19:43.432: g_converter_convert: assertion 'outbuf_size > 0' failed

(AppImageLauncher:26512): Gtk-CRITICAL **: 18:19:43.432: Unable to load resource for composite template for type 'GtkDialog': The resource at “/org/gtk/libgtk/ui/gtkdialog.ui” failed to decompress

(AppImageLauncher:26512): Gtk-CRITICAL **: 18:19:43.432: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26512): Gtk-CRITICAL **: 18:19:43.432: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26512): Gtk-CRITICAL **: 18:19:43.432: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26512): Gtk-CRITICAL **: 18:19:43.432: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26512): Gtk-CRITICAL **: 18:19:43.432: gtk_widget_class_bind_template_callback_full: assertion 'widget_class->priv->template != NULL' failed
Failed to connect to bus: Operation not permitted
Failed to connect to bus: Operation not permitted
Failed to connect to bus: Operation not permitted
Failed to connect to bus: Operation not permitted
```

Successful upload using docker: https://linux-hardware.org/?probe=46ce32a2fc

bsdhw, 2020-08-07:
Seems that this is a bug in your local AppImageLauncher, not in AppImage itself. AppImage doesn't use Gtk.

Please submit a bug to https://github.com/TheAssassin/AppImageLauncher

TheAssassin, 2020-10-28:
These GTK messages are just warnings, they're meaningless to the execution. The issue appears to be related to appimagelauncherfs being in some limbo state.

Just update to AppImageLauncher 2.2.0+, where the FUSE filesystem has been replaced with a more robust approach, and it'll work again.

Surendrajat, 2020-10-28:
@TheAssassin Thanks. I updated it and it's working fine, although I'm still getting these warnings.

```
$ sudo -E ~/Downloads/hw-probe-1.5-149-x86_64.AppImage -all -upload                                                                                              
QStandardPaths: wrong ownership on runtime directory /run/user/1000, 1000 instead of 0
QStandardPaths: wrong ownership on runtime directory /run/user/1000, 1000 instead of 0

(AppImageLauncher:10808): GLib-GIO-CRITICAL **: 17:27:14.220: g_converter_convert: assertion 'outbuf_size > 0' failed

(AppImageLauncher:10808): Gtk-CRITICAL **: 17:27:14.220: Unable to load resource for composite template for type 'GtkDialog': The resource at “/org/gtk/libgtk/ui/gtkdialog.ui” failed to decompress

(AppImageLauncher:10808): Gtk-CRITICAL **: 17:27:14.220: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:10808): Gtk-CRITICAL **: 17:27:14.220: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:10808): Gtk-CRITICAL **: 17:27:14.220: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:10808): Gtk-CRITICAL **: 17:27:14.220: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:10808): Gtk-CRITICAL **: 17:27:14.220: gtk_widget_class_bind_template_callback_full: assertion 'widget_class->priv->template != NULL' failed
Failed to connect to bus: Operation not permitted
Failed to connect to bus: Operation not permitted
Probe for hardware ... Ok
Reading logs ... Ok
Uploaded to DB, Thank you!

Probe URL: https://linux-hardware.org/?probe=7401cec82c
```

closed 2020-10-28 by Surendrajat

TheAssassin, 2020-10-28:
You get these warnings because GTK is pretty annoying when Qt apps run on it. All this theming is just broken by design. But, as said, you can just ignore those.