linuxhw/hw-probe #69: Wrong Computer Identification

almir-campos, 2020-09-01:
hw-probe identified my computer as a MacBookPro 10, 2, 2020. However it's a MacBook Pro 13 2012 2.5GHz – a1425. I'm running Ubuntu 20.04 / focal.

aadrian, 2020-09-01:
@almir-campos "MacBookPro10,2" it's the "model identifier" . If you search it, you will find exactly your description https://everymac.com/systems/apple/macbook_pro/specs/macbook-pro-core-i5-2.5-13-late-2012-retina-display-specs.html

almir-campos, 2020-09-01:
@aadrian You're right! :) Next question: what does it mean the "Year: 2020"  on the report?

bsdhw, 2020-09-02:
The only way to detect manufacture date of the laptop is to get it from the BIOS version. And this doesn't work properly if you've updated the BIOS.

Your BIOS is 284.0.0.0.0 04/12/2020. So the best estimation of the manufacture date is 2020 in your case.

almir-campos, 2020-09-02:
Cool. Thx!

closed 2020-09-02 by almir-campos