linuxhw/hw-probe #71: Please add "-m" to your inxi command inside your hw-probe Perl script

sebma, 2020-09-06:
Hi,
In your `inxi` command inside your `hw-probe` Perl script, you are missing memory info : 
```perl
$ grep "inxi -Fxx" $(which hw-probe)
            $Inxi = runCmd("inxi -Fxxxz --no-host 2>&1");
```
Can you please add `-m` to your `inxi` command inside your `hw-probe` Perl script ?

linuxhw, 2020-12-02:
Fixed by commit c8a19268576a8f1c20279a185443ca87c1a5f8fd

closed 2020-12-02 by linuxhw

sebma, 2021-08-08:
@linuxhw Thanks man. BTW : Why do you use a vicious smile tux as an avatar, do you plan to sell the data collected on https://linux-hardware.org ?

linuxhw, 2021-08-09:
@sebma The collected data is publicly available in multiple repositories: https://github.com/linuxhw?tab=repositories

The icon was introduced by original authors of the site in 2005 and saved for historical reasons: http://web.archive.org/web/20051024143339/http://linux-hardware.org/

sebma, 2021-08-09:
@linuxhw Oh I see. I forgot this collected data was anonymous.
Thanks for your explanation.
Keep up the good work.
God bless you.