linuxhw/hw-probe #75: Why is link to probe not persistent?

mimugmail, 2020-10-06:
Hi,

I'm playing around with hw-probe on OPNsense Firewall (FreeBSD/HBSD), maybe building a plugin for it.
Not really familiar with the tool but my first guess was, that when I execute `hw-probe -all -upload` the link I receive is always the same, but in fact it's everytime a different link.

Is this intended behaviour? 

Don't get me wrong, no critisicm, I just need to know it cause it depends the way I'd build a plugin with a nice UI :)

lvc, 2020-10-06:
Hi!

It creates a snapshot of hardware state at a specific point in time. So they are always different.

mimugmail, 2020-10-06:
Ok, thanks for this one, seems my test hardware was successfully collected. 
https://bsd-hardware.info/index.php?probe=05be956e1e

closed 2020-10-06 by mimugmail