linuxhw/hw-probe #76: Detected incorrect DE

bjeanes, 2020-10-08:
I just did the following probe: https://linux-hardware.org/index.php?probe=95485c3f4f

It detects my DE as `Deepin (X11) - GDM` when in fact it is GNOME (3.36) under Wayland. I _do_ however have Deepin screenshot tool installed, so there are some deepin dependencies on my system but in no way am I running either X11 (other than XWayland) nor Deepin as part of my desktop environment.

Let me know if there's any info I can give you from my system to help improve this check.

lvc, 2020-10-08:
Hi!

Please post your $XDG_SESSION_TYPE and $XDG_CURRENT_DESKTOP here.

bjeanes, 2020-10-08:
```
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=x11
```

OK that's weird. I just set up this machine a few weeks ago and specifically set it up with Wayland as per my laptop.

I can't account for why it's now running as (or reporting as) X11, other than perhaps my recent GNOME 3.36->3.38 upgrade is causing Wayland to fail to start, but can see that it was changed 3 days ago right after a boot, thanks to `snapper` taking snapshots at boot and each package change.

<details>
<summary>off-topic exploration of why my session reverted to X11</summary>

``` sh-session
❯ md5sum */snapshot/etc/gdm/custom.conf | awk '{ print $2 " " $1 }' | sort -n
[snip]
180/snapshot/etc/gdm/custom.conf 724f8bc88ffce55a889c84ebc3b8cba5
181/snapshot/etc/gdm/custom.conf 724f8bc88ffce55a889c84ebc3b8cba5
182/snapshot/etc/gdm/custom.conf ed0a2d19ba7344faa0feadd783bb4fa5
183/snapshot/etc/gdm/custom.conf ed0a2d19ba7344faa0feadd783bb4fa5
[snip]

❯ snapper diff 181..182 /etc/gdm/custom.conf
--- /.snapshots/181/snapshot/etc/gdm/custom.conf	2020-09-26 22:33:27.564670392 +1000
+++ /.snapshots/182/snapshot/etc/gdm/custom.conf	2020-10-06 11:27:18.844585171 +1100
@@ -1,19 +1,2 @@
-# GDM configuration storage
-
 [daemon]
-# Uncomment the line below to force the login screen to use Xorg
-#WaylandEnable=false
-
-[security]
-
-[xdmcp]
-
-[chooser]
-
-[debug]
-# Uncomment the line below to turn on debugging
-#Enable=true
-
+WaylandEnable=false

❯ snapper list --columns number,date,type,description
[snip]
179  | Tue 06 Oct 2020 10:54:15 | pre    | pacman -U --noconfirm --config /etc/pacman.conf -- /home/bjeanes/.cache/...
180  | Tue 06 Oct 2020 10:54:16 | post   | 1password-bin neovim-nightly visual-studio-code-bin                        
181  | Tue 06 Oct 2020 11:27:18 | single | boot                                                                       
182  | Tue 06 Oct 2020 11:28:14 | pre    | pacman -R -s --config /etc/pacman.conf -- spotify                          
183  | Tue 06 Oct 2020 11:28:14 | post   | libcurl-gnutls spotify                                                     
[snip]
```

So sometime between the boot at 11:27:18 and me installing a package right after I logged in less than 1 minute later, that file was rewritten. I wonder if that was my first boot into GNOME 3.38 and it rewrote that file after a failed Wayland start.
</details>

Either way, it is still reporting Deepin where it should report GNOME. It appears correct about X11 though! My apologies for not double checking my assumption there.

lvc, 2020-10-09:
Did you use `sudo -E hw-probe -all -upload` command (with `-E` option of `sudo`)?

bjeanes, 2020-10-09:
Yes. I copied and pasted the command from https://linux-hardware.org/index.php?view=howto

lvc, 2020-10-09:
Yep. This is a bug. And it's already fixed in master.

bjeanes, 2020-10-09:
Do you need me to test it or shall we just close the issue?

lvc, 2020-10-10:
It would be great if you could test. You need to replace the script `/usr/bin/hw-probe` by https://github.com/linuxhw/hw-probe/blob/master/hw-probe.pl

Thank you.

bjeanes, 2020-10-12:
Yup that appeared to work - https://linux-hardware.org/?probe=0746d41dd1

closed 2020-10-12 by bjeanes

taklert, 2020-10-12:
I think the issue is still there..

https://linux-hardware.org/?probe=aa0a4ef80c

Here is my hw-probe output..

And here's the truth:

https://imgur.com/a/IfuRv5y

linuxhw, 2020-10-12:
> I think the issue is still there..

Did you update it to master?