linuxhw/hw-probe #77: Misleading probe for Desktop Environments on Arch Linux (??)

itstoast27, 2020-11-07:
I had uploaded my probe to the website, and noticed that my Desktop Environment was listed as Deepin, when I had previously uninstalled Deepin from my installation, and replaced it with Xfce4. Maybe there's just a quirk with residual files on Arch? I wouldn't know. I'll be here for questions if need be.

bsdhw, 2020-11-08:
Please share your probe ID to debug.

itstoast27, 2020-11-08:
> Please share your probe ID to debug.

8ecf8a2eef

bsdhw, 2020-11-11:
Did you use `sudo -E hw-probe -all -upload` (with `-E` flag of sudo)?

itstoast27, 2020-11-11:
> Did you use `sudo -E hw-probe -all -upload` (with `-E` flag of sudo)?

I had copied it directly from the "Trends",page, which has the -E flag on it, so I presume that I did.

itstoast27, 2020-11-11:
Hmm, I can't seem to find any references of hw-probe anywhere in my ~/bash_history files, for both the root user and myself. I'm going to run the script again, ensuring that I use the -E flag, and see if any differences arise.

itstoast27, 2020-11-11:
Alright; new probe at 7b4eb00ecd; seems to have the same issue of proclaiming that my desktop environment is Deepin.

linuxhw, 2020-11-13:
Could you please provide:

    sudo -E echo $XDG_CURRENT_DESKTOP
    sudo -E echo $DESKTOP_SESSION
    hw-probe --dumpversion

itstoast27, 2020-11-13:
For reference, I had switched my Desktop back to LXQt because of storage space constraints on my system, and had uninstalled the "Deepin Archiver", and it seems to have fixed itself after doing another probe.  I didn't have internet access at the time, but my inspection of the output file seems to show that it's fixed. The probe is [here]https://mega.nz/file/9TYxFA7Y#d7M6f3lRMoFRPIJqRDSszIKuHXU9vH4W2ZoCv7FXw_4(url). Maybe my installation of Deepin's Archival program had tricked the hw-probe app? Hopefully you can make sense of this, and maybe recreate it.

itstoast27, 2020-11-13:
Oh, and the outputs of those commands, whoops.
```[karkat@karmiarch ~]$ sudo -E echo $XDG_CURRENT_DESKTOP
LXQt
[karkat@karmiarch ~]$ sudo -E echo $DESKTOP_SESSION
lxqt```

linuxhw, 2020-11-14:
Fixed on server. Please try again to verify.

itstoast27, 2020-11-18:
Sorry for the late response, but it seems to be fixed, at least after having switched to LXQt. The probe is at here[https://linux-hardware.org/?probe=6745b70f9d](url).

linuxhw, 2020-11-23:
Good to know! Thank you!

closed 2020-11-23 by linuxhw