linuxhw/hw-probe #80: Add exceptions or hide some information

alejandrobyte, 2020-12-12:
Hello,
Thank you very much for this software, it's great.
But I need hide some private information, how can we add masks or exceptions?
Thanks

linuxhw, 2020-12-14:
Hello,

First try recent version of hw-probe from master branch. Probably the problem has already been resolved.

Two ways to hide unwanted info from your probe:

1. Tell me via [E-mail](https://linux-hardware.org/index.php?view=contacts) what info should be hidden from your probe and I'll fix the hw-probe appropriately
2. Make a probe in three steps: a) `hw-probe -all`, b) edit `/root/HW_PROBE/LATEST/hw.info`, c) `hw-probe -upload`

Thanks!

alejandrobyte, 2020-12-14:
I'm talking to some hardware vendors, and even though they are listed as "Manufacturer" on the motherboard, they don't want any details of the "SubVendor: " ...,  appear.
It would be great to add terms to hide.
Thanks,