linuxhw/hw-probe #83: efibootmgr Log may contain UUIDs of partitions

Scrumplex, 2021-01-03:
If EFISTUB is used to boot a machine, the UUID of the machine is visible in `efibootmgr`. As it's UTF-16 or something the UUID is not clearly identified.

Example:

```
BootCurrent: 0002
Timeout: 1 seconds
BootOrder: 0002,0000,0003,0001,0004,0005,0006
Boot0000* Linux Boot Manager	HD(1,GPT,HASH_77F-D432-E137-BB9E-0E17B287E80F,0x800,0xfa000)/File(\EFI\SYSTEMD\SYSTEMD-BOOTX64.EFI)
Boot0002* Arch	HD(1,GPT,HASH_77F-D432-E137-BB9E-0E17B287E80F,0x800,0xfa000)/File(\VMLINUZ-LINUX-ZEN)r.o.o.t.=.U.U.I.D.=.X.X.X.X.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.X.X.X.X.X.X.X.X. .r.w. .q.u.i.e.t. .l.o.g.l.e.v.e.l.=.3. .v.t...g.l.o.b.a.l._.c.u.r.s.o.r._.d.e.f.a.u.l.t.=.0. .u.d.e.v...l.o.g._.p.r.i.o.r.i.t.y.=.3. .a.m.d.g.p.u...g.p.u._.r.e.c.o.v.e.r.y.=.1. .r.e.s.u.m.e.=.U.U.I.D.=.X.X.X.X.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.X.X.X.X.X.X.X.X. .r.e.s.u.m.e._.o.f.f.s.e.t.=.5.7.3.4.4. .s.y.s.r.q._.a.l.w.a.y.s._.e.n.a.b.l.e.d.=.1. .i.n.i.t.r.d.=.\.i.n.i.t.r.a.m.f.s.-.l.i.n.u.x.-.z.e.n...i.m.g.
Boot0003* UEFI OS	HD(1,GPT,HASH_77F-D432-E137-BB9E-0E17B287E80F,0x800,0xfa000)/File(\EFI\BOOT\BOOTX64.EFI)..BO
Boot0004* UEFI:CD/DVD Drive	BBS(129,,0x0)
Boot0005* UEFI:Removable Device	BBS(130,,0x0)
Boot0006* UEFI:Network Device	BBS(131,,0x0)

```

Scrumplex, 2021-01-03:
A potential fix could be adding the parameter `-u | --unicode | --UCS-2` which, as per man-page, "Handle[s] extra command line arguments as UCS-2 (default is ASCII)". That would remove the dots in between the letters. But this might cause other issue? I don't really know

linuxhw, 2021-01-04:
UUIDs prefixed by `HASH_` are not real. See https://github.com/linuxhw/hw-probe#privacy. UUIDs are decorated in the same way as MACs and serials, but formatted like regular UUIDs in order to save readability of logs.

Scrumplex, 2021-01-04:
Oh I wasn't really clear in my report.
 
The issue appears in this line:

```
Boot0002* Arch	HD(1,GPT,HASH_77F-D432-E137-BB9E-0E17B287E80F,0x800,0xfa000)/File(\VMLINUZ-LINUX-ZEN)r.o.o.t.=.U.U.I.D.=.X.X.X.X.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.X.X.X.X.X.X.X.X. .r.w. .q.u.i.e.t. .l.o.g.l.e.v.e.l.=.3. .v.t...g.l.o.b.a.l._.c.u.r.s.o.r._.d.e.f.a.u.l.t.=.0. .u.d.e.v...l.o.g._.p.r.i.o.r.i.t.y.=.3. .a.m.d.g.p.u...g.p.u._.r.e.c.o.v.e.r.y.=.1. .r.e.s.u.m.e.=.U.U.I.D.=.X.X.X.X.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.-.X.X.X.X.X.X.X.X.X.X.X.X. .r.e.s.u.m.e._.o.f.f.s.e.t.=.5.7.3.4.4. .s.y.s.r.q._.a.l.w.a.y.s._.e.n.a.b.l.e.d.=.1. .i.n.i.t.r.d.=.\.i.n.i.t.r.a.m.f.s.-.l.i.n.u.x.-.z.e.n...i.m.g.
```

I replaced my own UUID with Xs. Look at the args `r.o.o.t.=.U.U.I.D.=.<uuid>`

linuxhw, 2021-01-04:
Fixed by 0f61f0a8dfd7a33dbb07d101c48e67e8c2b084ee

linuxhw, 2021-01-04:
Thanks a lot!

closed 2021-01-04 by linuxhw