linuxhw/hw-probe #84: Canonical URLs for probes

grahamperrin, 2021-01-23:
Please

The initial URL, given to the user immediately after an upload, should be no different from the URL that's given when the probe is subsequently found by browsing or searching. 

### Examples

Initial URLs: 

* https://bsd-hardware.info/?probe=427bb18c90
* https://bsd-hardware.info/?probe=60d9540d35
* https://bsd-hardware.info/?probe=97a3ac7e36
* https://bsd-hardware.info/?probe=fcedf7a28d
* https://bsd-hardware.info/?probe=86e5ba4c5b
* https://bsd-hardware.info/?probe=e9c5982872

Subsequently found probes include: 

* https://bsd-hardware.info/index.php?probe=427bb18c90
* https://bsd-hardware.info/index.php?probe=60d9540d35
* https://bsd-hardware.info/index.php?probe=97a3ac7e36
* https://bsd-hardware.info/index.php?probe=86e5ba4c5b

… and so on. 

### Thoughts

I prefer to treat `index.php` as superfluous. Prefer terse URLs¸ as initially given. 

If that will be not easy to implement, then the initial URL should include `index.php`

Thanks

linuxhw, 2021-01-23:
Good idea! Thank you. This will be implemented ASAP.

grahamperrin, 2021-01-23:
Thank you! 

For now I'm using https://github.com/einaregilsson/Redirector with a wildcard include pattern: 

`https://bsd-hardware.info/*?probe=*`

– redirecting to: 

`https://bsd-hardware.info/?probe=$2`

linuxhw, 2021-01-26:
Fixed to be `https://bsd-hardware.info/?probe=XXX` everywhere.

closed 2021-01-26 by linuxhw