linuxhw/hw-probe #85: Consider presenting date and time of the probe, near the head of the page

grahamperrin, 2021-01-23:
Please, can you show the date and time?

It can be found in logs, but that's not as user-friendly as presenting it (for example) near the head of the main page. 

As a workaround I have been using Diigo to date some of my probes, for example: 

[image]

[image]

grahamperrin, 2021-04-02:
Gentle bump …

linuxhw, 2021-04-02:
Working on it ...

linuxhw, 2021-04-02:
Added a tooltip for the title. Have not found a better place for it on the page yet to not to overload it.

[image]

closed 2021-05-04 by linuxhw