linuxhw/hw-probe #90: PRIVACY VIOLATION: hw-probe on FreeBSD did upload wthout `-upload` option being given on command line

SlySven, 2021-02-24:
I ran the command as `root` without *any* options - yes, stupid I know now - on my mutli-OS box when it was running FreeBSD and the data was uploaded without my consent. If you are going to default to that on FreeBSD (could not see a repository for `hw-probe` on https://github.com/bsdhw but it seems to be linked to this repo) or any OS for that matter then you must ask for consent first!

:angry:

linuxhw, 2021-02-26:
Already fixed in master by the commit ab7b4b70785b9920488f2c6bc0c03c2ee418a1f0 5 months ago.

linuxhw, 2021-02-26:
Please wait for next tag release to adopt in FreeBSD.

closed 2021-02-26 by linuxhw

SlySven, 2021-02-26:
> Please wait for next tag release to adopt in FreeBSD.

Do you mean **don't use this on FreeBSD until the next version is available**? Well, that is not any good given that the existing package is already there in FreeBSD or are you saying that the package maintainer on FreeBSD should delete their port until they can update it - or that they should hold off on updating until you have the next release done? I suppose they *might* be able to backport just that change in that commit for the moment.

:confused:

linuxhw, 2021-02-27:
They can update it right now to the master version (like this is done by Manjaro). Master version is stable enough for this. Adding a patch is also a good solution.