linuxhw/hw-probe #94: Appimage offline mode

Moltkex, 2021-03-07:
Hi, 
I downloaded **hw-probe-1.5-164-x86_64.AppImage** and wanted to try and see if the offline mode as described here https://github.com/linuxhw/hw-probe#offline-view worked but got `ERROR: requires perl-Data-Dumper module` I installed `libdata-dumper-simple-perl` to no avail. This is on Debian testing. Is it possible with the appimage? Or is it only possible after installing hw-probe? Thanks.

linuxhw, 2021-03-07:
Hi,

You don't need `libdata-dumper-simple-perl`. The Data-Dumper module is included to Perl base package in Debian.

Probably we need to add this module to the appimage for this option to work.

Thanks.

Moltkex, 2021-03-07:
> Hi,
> 
> You don't need `libdata-dumper-simple-perl`. The Data-Dumper module is included to Perl base package in Debian.
> 
> Probably we need to add this module to the appimage for this option to work.
> 
> Thanks.

Ah, I see.  So, just to clarify, at this very moment, the offline mode doesn't work with the appimage? 
Thanks for your reply.

linuxhw, 2021-03-07:
Hm, the Data-Dumper module is already included to the appimage... Looks like incorrect installation of this module.

linuxhw, 2021-03-07:
> Ah, I see. So, just to clarify, at this very moment, the offline mode doesn't work with the appimage?
> Thanks for your reply.

Yep. Install this deb package instead: http://ftp.br.debian.org/debian/pool/main/h/hw-probe/hw-probe_1.5-1_all.deb

Thanks.

Moltkex, 2021-03-07:
> Yep. Install this deb package instead: http://ftp.br.debian.org/debian/pool/main/h/hw-probe/hw-probe_1.5-1_all.deb
>
> Thanks.

It's available in Testing repos so I'll install using apt. 
Thanks

linuxhw, 2021-03-18:
I could not get it to work and turned off this function in the AppImage. Use other probe methods instead: https://github.com/linuxhw/hw-probe#install

Thank you.

closed 2021-03-18 by linuxhw

Moltkex, 2021-03-18:
> I could not get it to work and turned off this function in the AppImage. Use other probe methods instead: https://github.com/linuxhw/hw-probe#install
> 
> Thank you.

I did. I installed the pkg from testing repos and uploaded my probe already :grin: 
Thank you.