linuxhw/hw-probe #97: Privacy, sensitivity: disclosure of serial numbers and other unique information

grahamperrin, 2021-04-23:
Follow up to #45, #56 and #83

### dmesg -- display the system message buffer

https://man.freebsd.org/dmesg(8)

`https://bsd-hardware.info/?probe=⋯&log=dmesg` can disclose serial numbers of disks. 

For example, from someone else's probe, with `⋯` used by me to manually overwrite information that might be considered sensitive: 

```text
…
[...] ugen1.3: <SanDisk Ultra> at usbus1
[...] umass0 on uhub3
[...] umass0: <SanDisk Ultra, class 0/0, rev 2.10/1.00, addr 3> on usbus1
[...] umass0:  SCSI over Bulk-Only; quirks = 0x4001
[...] umass0:3:0: Attached to scbus3
[...] da0 at umass-sim0 bus 0 scbus3 target 0 lun 0
[...] da0: <SanDisk Ultra 1.00> Removable Direct Access SPC-4 SCSI device
[...] da0: Serial Number...
[...] da0: 40.000MB/s transfers
[...] da0: 59232MB (121307136 512 byte sectors)
[...] da0: quirks=0x2<NO_6_BYTE>
[...] ugen1.3: <SanDisk Ultra> at usbus1 (disconnected)
[...] umass0: at uhub3, port 2, addr 3 (disconnected)
[...] da0 at umass-sim0 bus 0 scbus3 target 0 lun 0
[...] da0: <SanDisk Ultra 1.00>  s/n ⋯ detached
[...] (da0:umass-sim0:0:0:0): Periph destroyed
[...] umass0: detached
[...] ugen1.3: <SanDisk Ultra> at usbus1
[...] umass0 on uhub3
[...] umass0: <SanDisk Ultra, class 0/0, rev 2.10/1.00, addr 3> on usbus1
[...] umass0:  SCSI over Bulk-Only; quirks = 0x4001
[...] umass0:3:0: Attached to scbus3
[...] da0 at umass-sim0 bus 0 scbus3 target 0 lun 0
[...] da0: <SanDisk Ultra 1.00> Removable Direct Access SPC-4 SCSI device
[...] da0: Serial Number...
[...] da0: 40.000MB/s transfers
[...] da0: 59232MB (121307136 512 byte sectors)
[...] da0: quirks=0x2<NO_6_BYTE>
[...] ugen1.3: <SanDisk Ultra> at usbus1 (disconnected)
[...] umass0: at uhub3, port 2, addr 3 (disconnected)
[...] da0 at umass-sim0 bus 0 scbus3 target 0 lun 0
[...] da0: <SanDisk Ultra 1.00>  s/n ⋯ detached
[...] (da0:umass-sim0:0:0:0): Periph destroyed
[...] umass0: detached
```

I have not looked closely at logs, to tell whether serial information is disclosed for things other than drives. 

Some types of USB camera, maybe? 

No serial in this example: 

`uaudio1: <Microsoft Microsoft LifeCam Cinema(TM), class 239/2, rev 2.00/1.05, addr 12> on usbus0`

– maybe not a good example. It is, I suspect, a relatively _old_ webcam (borrowed from an office). 

## Other unique information

There may be cases other than dmesg.

linuxhw, 2021-04-26:
Hi,

This is now fixed. Please verify.

Thanks a lot for your help!

closed 2021-05-04 by linuxhw

grahamperrin, 2021-05-05:
https://bsd-hardware.info/?probe=6a96e2c5b1&log=dmesg looks OK now, at a glance. Thanks!