linuxhw/hw-probe #98: GhostBSD system detection

grahamperrin, 2021-05-01:
https://bsd-hardware.info/?probe=4bfec050c1 for example, 

> GhostBSD 20.04.02

– however this was a fresh installation, from `GhostBSD-21.04.27.iso` via https://www.ghostbsd.org/download where it's described as: 

> GhostBSD 21.04.27

https://github.com/ghostbsd/ghostbsd-src/blob/stable/13/sys/conf/package-version catches my eye, however the version there is `21.04.30` and stepping back through _blame_ at https://github.com/ghostbsd/ghostbsd-src/blame/stable/13/sys/conf/package-version does not find `21.04.27`

@ericbsd

linuxhw, 2021-05-04:
How can I find this version in the installed system?

ericbsd, 2021-05-04:
`pkg rquery '%v' ports`

grahamperrin, 2021-05-05:
Edge cases, the reported version can be superior to what's installed. 

For example, with `21.05.01`: 

[image]

* the second run of `pkg rquery '%v' ports` returned `21.05.04` _before_ updates to `21.05.04` were applied. 
 
I should treat this as negligible because in most such cases, the end user will receive an automated prompt to update.

ericbsd, 2021-05-05:
@grahamperrin It is not perfect, but it is the closes we have right now.

linuxhw, 2021-05-21:
Fixed by commit a4d3614b56889c6f3e08ec9dec82f71505bb2299

Please check.

ericbsd, 2021-12-18:
The 21.12.18 update is coming later today. There will /etc/os-release containing GhostBSD and GhostBSD version instead FreeBSD. Also, there was /bin/ghostbsd-version added on the last update.

ericbsd, 2021-12-18:
This is wrong:
```
> cat /root/HW_PROBE/LATEST/hw.info/host
arch:amd64
boot_mode:EFI
cores:16
current_desktop:helloDesktop
de:helloDesktop
display_server:X11
dual_boot:0
dual_boot_win:0
filesystem:zfs
freebsd_release:13.0
freebsd_version:13.0-STABLE
hwaddr:5a97ae9b318126e34f71f12be6ea447e
kernel:13.0-STABLE
lang:en_US.UTF-8
microarch:Zen+
model:X470 AORUS ULTRA GAMING-CF
monitors:1
nics:1
part_scheme:GPT
probe_ver:1.6
ram_total:67108864
ram_used:1905664
sockets:1
space_total:327
space_used:54
submodel:X470 AORUS ULTRA GAMING
subvendor:Gigabyte Technology Co., Ltd.
system:ghostbsd-21.12.05
system_version:21.12.05
threads:1
type:desktop
uuid:76EA6BBF-D8CD-1EBF-C433-33798A4D7173
vendor:Gigabyte Technology Co., Ltd.
video_memory:4
year:2021
```
de should be MATE.

ericbsd, 2021-12-18:
```
> env | grep mate
DESKTOP_SESSION=mate
XDG_SESSION_DESKTOP=mate
GDMSESSION=mate
```

ericbsd, 2021-12-18:
```
> env | grep MATE
XDG_CURRENT_DESKTOP=MATE
MATE_DESKTOP_SESSION_ID=this-is-deprecated
```

ericbsd, 2021-12-18:
I see my fault I forgot `-E` with sudo.

ericbsd, 2021-12-18:
PR  #114 will take care of this ticket.

danfe, 2022-01-24:
> Fixed by commit a4d3614

I guess I better comment here than to the commit itself. :-)

> Please check.

So now it calls `pkg rquery '%v' ports` unconditionally on all BSD systems, e.g. vanilla FreeBSD, not just on GhostBSD where it makes sense.  This could halt the probe for users of slow or filtered networks, because this command queries *remote* repository.  Consider the following patch:
```diff
-    if(checkCmd("pkg"))
+    if(checkCmd("pkg") and $Name eq "ghostbsd")
     {
         $GhostBSDPortsVer = runCmd("pkg rquery \'\%v\' ports");
```
> This is wrong:
> current_desktop:helloDesktop
> de:helloDesktop

This is because of `ps x | grep`, see #119.

closed 2022-02-03 by linuxhw

linuxhw, 2022-02-03:
Hi all!

I've merged both patches. Do you need a new tag (e.g. `1.6.2`)?

ericbsd, 2022-02-03:
> Hi all!
> 
> 
> 
> I've merged both patches. Do you need a new tag (e.g. `1.6.2`)?

That would be nice.

linuxhw, 2022-02-03:
Tagged https://github.com/linuxhw/hw-probe/releases/tag/1.6.2

danfe, 2022-02-04:
> The second run of `pkg rquery '%v' ports` returned `21.05.04` _before_ updates to `21.05.04` were applied.

If version of the locally installed package is needed, why call `pkg rquery` and not `pkg query`?

ericbsd, 2022-02-04:
> > The second run of `pkg rquery '%v' ports` returned `21.05.04` _before_ updates to `21.05.04` were applied.
> 
> 
> 
> If version of the locally installed package is needed, why call `pkg rquery` and not `pkg query`?

Do not worry about that. It is legacy code and will never run on GhostBSD since to install this packages you need to be at the latest and the latest use ghostbsd-version.

PR #114 (2022-02-03): Changed port GhostBSDPortsVer by GhostBSDVer to use ghostbsd-version
I tested the change here is the results:
```
/u/h/e/p/e/hw-probe > sudo -E perl hw-probe.pl -all 
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
 /u/h/e/p/e/hw-probe > cat /root/HW_PROBE/LATEST/hw.info/host
arch:amd64
boot_mode:EFI
cores:16
current_desktop:MATE
de:MATE
display_server:X11
dual_boot:0
dual_boot_win:0
filesystem:zfs
freebsd_release:13.0
freebsd_version:13.0-STABLE
hwaddr:5a97ae9b318126e34f71f12be6ea447e
kernel:13.0-STABLE
lang:en_US.UTF-8
microarch:Zen+
model:X470 A [... cut]