linuxhw/hw-probe #100: When run without flags it would default to uploading

py-crash, 2021-06-09:
If you run the software without any flags (in my case `flatpak run org.linux_hardware.hw-probe`) it would default to searching the hardware and uploading it. This could lead to an accidental leak of information, or at least to an unwanted result. 

The default behavior  should be just print a warning or print the help.

linuxhw, 2021-06-10:
To fix this, we need to distinguish two situations from inside the flatpak:

1. The probe is run by clicking on the desktop file
2. The probe is run by executing manually in terminal `flatpak run org.linux_hardware.hw-probe`

I have not found a way to do this yet. Please help with this.

sonnyp, 2021-11-15:
@linuxhw please address this - Flathub.org received a request to remove hw-probe because of it.

It shouldn't upload anything without user consent - no matter if it's from the desktop file, `flatpak run org.linux_hardware.hw-probe` or `hwprobe`.

https://discourse.flathub.org/t/please-remove-hardware-probe-it-is-not-a-desktop-application-and-has-surprising-network-activity/1803/9

sonnyp, 2021-11-15:
> I have not found a way to do this yet. Please help with this.

It's unclear to me - why do you need to differentiate ?

linuxhw, 2021-11-15:
> > I have not found a way to do this yet. Please help with this.
> 
> Why do you need to differentiate ?

This is due to bug https://gitlab.gnome.org/GNOME/gnome-software/issues/552

Running from the AppCenter just don't show anything where I can say "Yes, I confirm uploading of anonymized hardware info to the public database".

Can we just change the description of the app somehow to let people understand clearly how the app works while fixing this bug?

linuxhw, 2021-11-15:
I’ve described clearly how the app works in a new build: https://flathub.org/builds/#/builders/32/builds/67237

linuxhw, 2021-11-15:
> @linuxhw please address this - Flathub.org received a request to remove hw-probe because of it.
> 
> It shouldn't upload without user consent - no matter if it's from the desktop file, `flatpak run org.linux_hardware.hw-probe` or `hwprobe`.
> 
> https://discourse.flathub.org/t/please-remove-hardware-probe-it-is-not-a-desktop-application-and-has-surprising-network-activity/1803/9

How to omit this question when running by Launch button from the AppCenter? How it will behave if we will not answer?

sonnyp, 2021-11-16:
> How to omit this question when running by Launch button from the AppCenter?

Why would you?

linuxhw, 2021-11-16:
> > How to omit this question when running by Launch button from the AppCenter?
> 
> Why would you?

I'm not sure how it will behave in the background if not answering the question.

Forget it. I've started to implement the GUI app...

sonnyp, 2021-11-16:
The button shouldn't automatically upload without user consent. If that means the button doing nothing (at least until you have a GUI or until the flatpak/Software bug is fixed) so be it.

sonnyp, 2021-11-16:
We are discussing disallowing `<component type="desktop-application">` for CLI applications - you may be interested in joining the conversation https://discourse.flathub.org/t/flathub-needs-a-policy-about-command-line-tools/1855

linuxhw, 2021-11-18:
This is now fixed by adding the GUI.

closed 2021-11-18 by linuxhw