linuxhw/hw-probe #101: Review dependencies on FreeBSD

grahamperrin, 2021-06-11:
Reading <https://github.com/helloSystem/ISO/commit/dd9d82faf0a4f76fdae47c3c0c553b847f2e372f> alongside <https://www.freshports.org/sysutils/hw-probe/#history> and <https://www.freshports.org/sysutils/hw-probe/#dependencies>, I wonder whether the port to FreeBSD is missing dependencies such as these: 

* [sysutils/lsblk](https://www.freshports.org/sysutils/lsblk/)
* [sysutils/pciutils](https://www.freshports.org/sysutils/pciutils/)
* [sysutils/usbutils](https://www.freshports.org/sysutils/usbutils/)

If so: should this be fixed here in GitHub, or in sysutils/hw-probe downstream?

linuxhw, 2021-06-11:
This should be done downstream: https://bugs.freebsd.org/bugzilla/

We don't require these packages in upstream to support all FreeBSD versions down to 6.x where such packages don't exist.

grahamperrin, 2021-07-16:
Thanks, 

> This should be done downstream:

Updated: FreeBSD bug [252282 – sysutils/hw-probe runtime dependencies](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=252282)

closed 2021-07-16 by grahamperrin