linuxhw/hw-probe #103: FreeBSD GENERIC kernel always(ish) detected as Pine Rockpro64

agrajag9, 2021-07-14:
This is a SolidRun Honeycomb: https://bsd-hardware.info/?probe=647138144b

We can easily see here how this happened:

https://github.com/linuxhw/hw-probe/blob/f450cad0cf686756c32689595e6d0092933e5959/hw-probe.pl#L3878-L3883

```
$ kldstat -v | grep -i rockchip_dwmmc
                313 rockchip_dwmmc/mmc
                312 ofwbus/rockchip_dwmmc
                311 simplebus/rockchip_dwmmc
```
And in the GENERIC kernconf:

https://github.com/freebsd/freebsd-src/blob/29c8295312dc5eef8a0573bccdacd56167859cb2/sys/arm64/conf/GENERIC#L376

Perhaps we want to use something from the ACPI tables instead, when available? @jnettlet may have more to say about the best way to determine Vendor and Model though on systems like this.

evadot, 2021-07-14:
We have a kenv "kern.cfg.order" to select between booting in acpi or fdt but I'm not sure that we have something to get this info at runtime.
Also dmidecode doesn't seems to work on my honeycomb here, something I'll have to look at ...

linuxhw, 2021-07-16:
Disabling these heuristics temporarily.

agrajag9, 2021-08-03:
For ACPI-based systems, it turned out to be an issue with not reading SMBIOS3 magic properly and, at least on SolidRun Honeycomb, is fixed by https://reviews.freebsd.org/D28737

I cannot speak to DeviceTree-based systems and will leave that to evadot.