linuxhw/hw-probe #104: feature request: average temperature / fanspeed should also be apart of the test somehow

andrewfenn, 2021-07-22:
This is specifically about notebooks and having a good repository of knowledge on seeing how well temperature and fanspeed is being handled for different devices. What distro is handling it better, etc. I understand it wouldn't be exact data but having a large data volume set of temperatures along with fan speed configurations would be helpful to identifying issues of devices not being handled properly.

linuxhw, 2021-08-06:
How about lm_sensors?

https://linux-hardware.org/?probe=3600f5a7a8&log=sensors

andrewfenn, 2021-08-09:
This looks great. Then I guess what I am requesting since that information is already available is for there to be some trend on the website to be able to see differences in temperatures for the same hardware between different upgrades / distros. Do you think it would be possible / helpful to do?

linuxhw, 2021-08-10:
Need to think. But usually people don't create more than one or max two probes of the same hardware on average. Comparing different instances of the same hardware is not correct too due to different operating conditions.

Also the temperature of my personal laptop linearly depends on the temperature outside the window.

andrewfenn, 2021-08-11:
Yes, I think it's not going to be super accurate. I too have been thinking about how to handle this. I was thinking more along the lines of how well controlled fan-speeds are across different setups to be able to highlight what is best controlling notebooks if that makes more sense.

linuxhw, 2021-08-11:
Please share your thoughts on possible metrics to implement. One of the metrics may be minimal/average/maximal temperature per model to find out coolest/hottest models on the market.

linuxhw, 2021-10-30:
The repository with all lm_sensors reports has been created: https://github.com/linuxhw/Sensors

But analysis report (see README.md) is not ready yet - just showing minimal reached temp per model.