linuxhw/hw-probe #107: linuxhw/ but could it work on macOS?

alexmyczko, 2021-09-21:
just being curious... i like the stats part, and i am aware it is for hatdware support. on the other hand it delivers also stats like popcon.debian.org, which popcon.ubuntu.com also should but theirs is broken aince 5 years. there also is the much less popular linuxcounter.

linuxhw, 2021-09-23:
As you know, MacOS is sold together with hardware. So every hardware configuration is supported and therefore we don't need a database of supported/unsupported hardware. But it is still possible to create a database in degenerate form to store reliable / unreliable configurations (taking into account broken drives, WiFi cards, batteries, etc.). The hardware probe tool will operate just like a new hardware lister on MacOS.

We can host MacOS probes on https://bsd-hardware.info the same way as for all other BSD derivatives.

I will gladly do it if there are interested people who can give me a new MacBook ;)

alexmyczko, 2021-09-27:
```
perl hw-probe.pl -all -upload
mkdir /root: Read-only file system at hw-probe.pl line 17031.
```

closed 2022-06-19 by alexmyczko

Low-power, 2022-11-01:
After adding `darwin` to `@KNOWN_BSD`:
```
# uname -a
Darwin x 10.3.0 Darwin Kernel Version 10.3.0: Fri Feb 26 11:58:09 PST 2010; root:xnu-1504.3.12~1/RELEASE_I386 i386
# perl hw-probe.pl -all -maximal -upload 
WARNING: 'dmidecode' package is not installed
Probe for hardware ... Ok
Reading logs ... invalid option or syntax: -b
top usage: top
		[-a | -d | -e | -c <mode>]
		[-F | -f]
		[-h]
		[-i <interval>]
		[-l <samples>]
		[-ncols <columns>]
		[-o <key>] [-O <secondaryKey>]
		[-R | -r]
		[-S]
		[-s <delay>]
		[-n <nprocs>]
		[-stats <key(s)>]
		[-pid <processid>]
		[-user <username>]
		[-U <username>]
		[-u]

Ok
WARNING: failed to detect EDID
ERROR: not enough info collected
ERROR: failed to upload probe
```

sebma, 2023-06-18:
@Low-power `dmidecode` can be installed using [MacPorts](https://ports.macports.org/) : `sudo port install dmidecode`