linuxhw/hw-probe #110: PRIVACY: hw-probe uploads without user consent

sonnyp, 2021-11-15:
https://discourse.flathub.org/t/please-remove-hardware-probe-it-is-not-a-desktop-application-and-has-surprising-network-activity/1803

Please do not upload without user consent - no matter if it's from GUI, CLI or `flatpak run`

See also https://github.com/linuxhw/hw-probe/issues/90

sonnyp, 2021-11-15:
https://github.com/linuxhw/hw-probe/issues/100

closed 2021-11-15 by sonnyp

linuxhw, 2021-11-15:
Hello!

Thanks for reporting this!

I want to implement some user interaction since 2018 with no luck: https://gitlab.gnome.org/GNOME/gnome-software/issues/552

Need to fix this flatpak bug first. It's a blocker for such feature.

BTW: why people don't read the description of the package?

linuxhw, 2021-11-15:
I’ve described clearly how the app works in a new build: https://flathub.org/builds/#/builders/32/builds/67237

linuxhw, 2021-11-18:
Fixed by adding the GUI.