linuxhw/hw-probe #113: I uploaded three probes but when I click HWid I only see one

mailinglists35, 2021-12-03:
they differ in kernel dmesg (different kernels & kernel with BUG message)
I can view each probe indvidually but when I click the HWid I only see one probe
https://linux-hardware.org/?probe=e73cda5c1e
https://linux-hardware.org/?probe=115b1a128e
https://linux-hardware.org/?probe=74f451b21f
shouldn't I see all probes when I click HWid?

linuxhw, 2021-12-03:
Yes, you should, but after a delay. New probes are not approved immediately. We need some time to check for privacy violations and remove duplicates and spam. Also this delay is needed to revoke erroneously uploaded probes.

closed 2021-12-03 by linuxhw

linuxhw, 2021-12-03:
Although you can see your personal unapproved probes if using the [inventory ID](https://github.com/linuxhw/hw-probe#inventory).

mailinglists35, 2022-01-12:
hello again,

all my old probes show the log files with strikethrough font and they are not clickable.
I tried re-uploading a new probe https://linux-hardware.org/?probe=a80be6a29f

it worked for a while and now again the file names are strikethrough

Am I doing something wrong?
This is the command I used:

sudo ./hw-probe-1.5-149-x86_64.AppImage -all -upload

mailinglists35, 2022-01-12:
the machine exists for real, it's not spam. It has no vendor name / serials in dmidecode because it's a no-name chinese SoC

linuxhw, 2022-01-17:
> hello again,
>
> all my old probes show the log files with strikethrough font and they are not clickable. I tried re-uploading a new probe https://linux-hardware.org/?probe=a80be6a29f
>
> it worked for a [...]

This was due to another DDoS attack to the database. We experience such attacks ~once a month when people are mistakenly trying to download it through the web interface.