linuxhw/hw-probe #115: Web site banner/header versus in-page anchoring

grahamperrin, 2021-12-24:
## For example

1. https://bsd-hardware.info/?probe=539b95f535#usb:05ac-022a

## Expected

2. visibility of the row that includes the phrase `05ac:022a` – for the _touchpad_

## Actual result

2. invisibility (behind the banner)
3. visibility of the subsequent row – for a _monitor_

[image]

## Workaround

4. key <kbd>Up</kbd>

## Thoughts 

In-page anchors are, like, a hidden feature, but still: 

* it will be smart if generic page layout can be redesigned, for the banner to **never** hide what's required. 

Thanks

linuxhw, 2021-12-28:
Thank you for reporting this.

This is now fixed. Please check.

grahamperrin, 2021-12-28:
Tested in Firefox 95.0.2 on FreeBSD 14.0-CURRENT. Fix confirmed in this context. 

<https://bsd-hardware.info/?probe=539b95f535#pci:8086-284b-106b-00a1> brings into sight an _82801H (ICH8 Family) HD Audio Controller_. 

Thank you!

closed 2021-12-28 by grahamperrin