linuxhw/hw-probe #116: Strange use of "â€”" (without the leftmost and rightmost quotation marks) in README.md

Gomesz785, 2021-12-31:
In the file README.md from line 92, there is a starnge use of the following:
`â€”`

Probably same cause as this bug https://stackoverflow.com/questions/2477452/%c3%a2%e2%82%ac-showing-on-page-instead-of.

danfe, 2022-01-24:
I don't see that in master (or 1.6.1).  While `README.md` is indeed UTF-8, it's only because of the three mdash characters (e2 80 94).  If I do `sed 's,—,--,' README.md` the output would be pure ASCII.

Gomesz785, 2022-01-24:
> I don't see that in master (or 1.6.1). While `README.md` is indeed UTF-8, it's only because of the three mdash characters (e2 80 94). If I do `sed 's,—,--,' README.md` the output would be pure ASCII.

>In last week’s article, you learned a short process that solves most encoding problems. But there’s one encoding problem that’s much harder to solve.
>
>I know you’ve seen it. (Or maybe youâ€™ve seen it?) It’s when a curly quote turns into â€™, or an **em-dash turns into â€”.** It’ll make you think you’ve gone crazy. It should just work!

quoted from https://www.justinweiss.com/articles/how-to-get-from-theyre-to-theyre/

Yeah that should be the case, but now it looks gone.

closed 2022-01-24 by Gomesz785