linuxhw/hw-probe #119: Searching for running processes with "ps x | grep" is wrong

danfe, 2022-01-24:
Results would contain the `grep(1)` line itself (with arguments), so this construct will always find what's being searched.  Also, there's needless use of pipe.  Consider the following patch:
```diff
-        my $WMs = runCmd("ps x | grep wm");
+        my $WMs = runCmd("pgrep -l wm");
         if($WMs=~/\s(\w+wm)/) {
             $Sys{"Current_wm"} = $1;
         }
         
-        if(runCmd("ps x | grep start-hello"))
+        if(runCmd("pgrep -l start-hello"))
```
Now the probe correctly detects my IceWM instead of bogusly reporting that I run helloSystem DE.

linuxhw, 2022-02-03:
`pgrep -l start-hello` returns empty output on helloSystem.

[image]

linuxhw, 2022-02-03:
Fixed by commit 1274919ef786285bbdcf14329413d7545e68e433

closed 2022-02-03 by linuxhw

danfe, 2022-02-03:
That's because it's not part of the process name (`/bin/sh`) but its arguments.  For that you'd need to use `-f` switch.  From `man pgrep(1)`:

> -f                Match against full argument lists.  The default is to match against process names.

Committed fix is ugly. :-(